t2linux/T2-Ubuntu #1: Ubuntu doesn’t boot from ISO

alazarolop, 2020-04-27:
Hi, 

First of all, thank you for creating the Ubuntu iso. I’ve tried to use it in my MacBook Pro 15,3, but I wasn’t able to boot Ubuntu. 

With any of the boot options (try or directly install), I got a message of `dhd_os_wifi_platform_set_power: Setting wl_reg_on to 1 succeeded` and then the display turns black, though still on. After a minute the TouchBar turns on, but the display stays black. I’ve also tried to use `nosplash` option, but I don’t get any verbose message neither. 

If I use a default Ubuntu ISO, I can boot and complete the installation process using USB keyboard and mouse. 

Thank you

alazarolop, 2020-04-30:
Apparently, it could be related to a kernel issue in 5.6 that breaks `amdgpu` driver https://github.com/Dunedan/mbp-2016-linux/issues/142 .

marcosfad, 2020-05-03:
Please try the new release. That should be working now. There where errors on the Grub configuration. 
About the error, I see this too, but is loading afterwards. Enjoy.

alazarolop, 2020-05-04:
Ok. Thank you a lot!

alazarolop, 2020-05-06:
@marcosfad thank you for the effort and the latest improvements. 

Just for the records, the new 5.6.10 release didn't work out for my MBP 15,1 eventually, but a custom iso with the kernel 5.4.27-mbp did it. It's really awesome!

closed 2020-05-06 by alazarolop