t2linux/T2-Ubuntu #4: WiFi not working

vinodhmoodley, 2020-05-26:
I have the 2018 13" MacBook Pro with touch bar and the keyboard and track pad work fine. The internal SSD is also detected correctly. Thanks
However, WiFi isn't working and i'm not sure where to start trying to figure it out. Any help or suggestions will be appreciated.

gamunu, 2020-06-03:
@vinodhmoodley try packages from https://packages.aunali1.com/apple/wifi-fw/ . The firmware that came with Catalina did not work for me either.

vinodhmoodley, 2020-06-03:
> @vinodhmoodley try packages from https://packages.aunali1.com/apple/wifi-fw/ . The firmware that came with Catalina did not work for me either.

Thanks for trying to help but as mentioned in my post, I did use that Mojave firmware and not the Catalina version. In the mean time, I've installed Fedora and it did the same thing. I've re-installed Ubuntu several more times and it does the same thing. I asked for help on the Linux on MBP 2018+ Discord and one other user has the same issue as me.

vinodhmoodley, 2020-10-17:
I got the WiFi issue fixed. I used a newer build of your mbp-ubuntu iso and installed like normal. I then copied the firmware files exactly like what I did previously and now it works. I'm not sure what changed but somehow or other, it's working despite following the exact same method I used before.

However, if i try to connect to a WiFi network, any WiFi network, it keeps asking for a password even though the password I enter is definitely correct. Turns out that Network Manager needs IWD instead of WPA Supplicant to resolve this isssue. I installed IWD and edited the Network Manager config to use IWD as it's back end and it now works. 

I detailed what I did here if anyone else has this issue:
https://vinodhsblog.co.za/ubuntu-20-04-lts-on-a-macbook-pro/

closed 2020-10-17 by vinodhmoodley