t2linux/T2-Ubuntu #5: Can't expand livecd.zip, it's an unsupported format

MoonPadUSer, 2020-06-26:
This is the error message I get when I try to unzip the downloaded file from releases:
`Unable to expand "livecd.zip". It is in an unsupported format.`
I'm not sure if I'm just missing smth but it says I am supposed to download the iso from the releases, but since there is only this livecd.zip file (which probably contains the .iso file) I downloaded it, but now I can't expand it. Huh?

ceinem, 2020-06-26:
Yeah, I had the same problem on my Mac. Just use another zip tool instead of the default Mac tool. I for example used [The Unarchiver](https://apps.apple.com/de/app/the-unarchiver/id425424353?mt=12) which worked completely fine.

MoonPadUSer, 2020-06-26:
thank you a lot, I suggest putting this information inside of the readme. :D have a nice day

closed 2020-06-27 by MoonPadUSer