t2linux/T2-Ubuntu #7: Black screen

peryngveohlin666, 2020-06-29:
Tried the newest kernel with mbp 15,1, catalina 10.15.5. I get a black screen, the touchbar loads so the drivers for them are loading in so it is probably about the amd gpu. I have seen that it might been fixed on aunalis repo, would be a good idea to bring it to the ubuntu version as well. https://github.com/aunali1/linux-mbp-arch/blob/master/2001-drm-amd-display-Force-link_rate-as-LINK_RATE_RBR2-fo.patch

vainkop, 2020-07-08:
> Tried the newest kernel with mbp 15,1, catalina 10.15.5. I get a black screen, the touchbar loads so the drivers for them are loading in so it is probably about the amd gpu. I have seen that it [...]

I also get black screen on my A1990 itself, but the external display works ok.

On the other hand, both screens work ok on the regular Ubuntu 20.04 with 5.4 kernel including brightness & everything, probably because Intel video card is working instead of AMD.

Would be nice to be able to fix it!

brycefranzen, 2020-07-17:
I'm getting a black screen as well. I cannot boot into "live" or "install".  Touchbar is functional, but the screen is black.

peryngveohlin666, 2020-07-17:
yep, for people who want to know more in detail: apparently this patch breaks the amdgpu kernel module for the 16,1 model and it fixes it for the 15,1 model (if i am not wrong) and the maintainer for this repo has a 16,1 macbook pro so he uses the 16,1 version without the patch, you can either fork and maintain your own kernel and distro for 15,1 or try fedora/arch/manjaro versions of the mbp-linux kernel which all seem to be working fine with the mbp 15,1. I am going to close it as it is not an active issue with the repository.

brycefranzen, 2020-07-17:
:( The repo says it works on 2018+ models.  Since it doesn't work on 2019 A1990, I would say there is a problem with this repo.  Please re-open so it can be fixed.  We shouldn't have to maintain a separate kernel from the project for this to work :(

peryngveohlin666, 2020-07-17:
i will reopen it but you should probably discuss this on the discord server

vainkop, 2020-07-17:
I've fixed black screen by simply disabling amdgpu, but it's not off, so after a while laptops' fan start to be white loud as it starts to overheat.
```
echo "blacklist amdgpu" >> /etc/modprobe.d/blacklist.conf
update-initramfs -u
```
To check whether the driver is blacklisted or not you should run the following command, which should output nothing:
``lsmod | grep amdgpu```

marcosfad, 2020-07-18:
Hi everybody, thank you for using the distro an reporting bugs. I will update the build script to release 2 versions. One for the newer models (-mbp) and one for the older ones (mbp-alt). I hope this will work for most of us.
I'm working with Aunali1 to fix this issue. I hope that we soon have a unique kernel to cover both systems. Stay tune.

marcosfad, 2020-07-21:
New release process is there. Please try out with version works for you. Please post in this issue, your macbook model and the version that worked for you. This will be the workaround until we are able to fix the real issue. Look forward to your feedback

peryngveohlin666, 2020-07-21:
boots up but didn't try installing yet, did you also configure dkms and the kernel repository or is it going to break after updating the kernel?

marcosfad, 2020-07-21:
dkms and repo are already installed. You can update the kernel without any trouble and dkms should install the necessary extra drivers.

brycefranzen, 2020-07-22:
Boots up into Live mode!  However, after running the install and attempting to boot from the installed system, numerous errors are thrown and it doesn't boot :(.  I'll keep messing with it.

peryngveohlin666, 2020-07-26:
what exactly are the errors? did you get it to work?

marcosfad, 2020-07-27:
Hi! Please try always the mbp-alt version first. It seems to work in almost all models. If it is not working. Please write down wich model is yours.

peryngveohlin666, 2020-07-28:
I have successfully installed mbp-alt while i couldn't even boot the original one on macbook pro 15,1

luzhuomi, 2020-07-29:
I have tried both alt and non-alt on my macbook pro 15,1 2018 with vega, both failed.
For alt, can't even see the normal desktop after entering "trial". The desktop flickered then blank.
For alt, it advanced one step further. I am able to see the desktop and no blank screen. However when I try to move the mouse to click on the install icon, it is not responding then eventually all GUI become distorted and funny characters. I guess it has to do with the vega GPU. I tried on an external monitor also no go.

But thanks for the effort.

vainkop, 2020-07-31:
I've decided to completely reinstall the OS & https://github.com/marcosfad/mbp-ubuntu/releases/download/v20.04-5.7.9-1/livecd-5.7.9-mbp-alt.zip completely solved all my issues on MBP 15,1 2018 A1990 except for wifi, which I fixed later by adding correct firmware (https://packages.aunali1.com/apple/wifi-fw/18G2022/)
I have 2 external screens & the MBPs' screen, all 3 working ok.
The sound is working out of the box, although haven't tested the mic yet.
The touch bar & keyboard are working ok.

marcosfad, 2020-10-11:
I think we can close this issue now. Don't we?

closed 2020-10-11 by marcosfad