t2linux/T2-Ubuntu #8: Keyboard backlight not working

ceinem, 2020-07-07:
I'm running this on a MacBookPro16,1, mostly it works really well, with the known issues of WiFi and the Microphone. 
I wanted to know if anyone had issues with the keyboard backlight or got it working? 

It shows the options of changing the backlight brightness on the TouchBar, but they don't do anything in my case.

swinder0161, 2020-08-06:
Same issue with me on MBP-16,1, someone please help.

marcosfad, 2020-10-11:
hi, yes. Unfortunately this is also an issue of the 16,1 model. I haven't found I fix yet.

reynaldliu, 2021-04-11:
HDMI external monitor
`command +p` and push `enter `
Then unplug it, you can restore the display

Or sometimes long press the Power button to turn on the screen for 2 seconds

AdityaGarg8, 2023-02-10:
The driver for the backlight is now available in the latest kernels

closed 2023-02-10 by AdityaGarg8