t2linux/T2-Ubuntu #9: Touchbar not working after suspend

MoonPadUSer, 2020-07-08:
I am on a 15,4 and after being suspended once the touchbar is not working anymore.
The `apple_ib_tb` module still shows up in `lsmod` but it doesn't seem to work

marcosfad, 2020-10-11:
Hi, 
For that I added this small script:
https://github.com/marcosfad/mbp-ubuntu/blob/master/files/suspend/rmmod_tb.sh
Check out here:
https://github.com/marcosfad/mbp-ubuntu/blob/master/01_build_file_system.sh#L39
It supposed to remove the driver and added later again. 

You may try this manually when the problem come up.

reynaldliu, 2021-04-07:
mbp 16,1 same

tpchuckles, 2021-07-21:
Any way to get this to run on lid close? might be a timing issue. (manually clicking "suspend" and then waiting for everything to go dark, then closing the lid, seems to work. just closing the lid means i won't get the touchbar once we're back up)

alternatively, any commands to get the touchbar back once it's gone? ("modprobe -r apple_ic_tb ; modprobe apple_ic_tb" doesn't seem to work. as though crashing the module means we can't restart it or something...)

AdityaGarg8, 2023-02-10:
Bug has been fixed in the latest kernels

closed 2023-02-10 by AdityaGarg8

AdityaGarg8, 2025-12-17:
Are you using tiny-dfr or normal touchbar?

jondkelley, 2026-01-22:
Any updates? Is anyone taking ownership of this?