t2linux/T2-Ubuntu #11: [MBP-16] Cannot Resume after suspend

swinder0161, 2020-08-06:
One I suspend my machine, cannot resume back.

I am not sure if suspend is even going through or not, the screen goes black and I feel no fan or other vibration also.

Is it same issue as the author has written that shutdown hangs?

Anyone else facing the same issue?

marcosfad, 2020-10-11:
Hi, we are all experiencing the same issue. There is no fix for that at the moment.
I can recommend you to join the discord channel.

reynaldliu, 2021-04-07:
i am same

doublebyte1, 2021-06-18:
Hi ! I am also experiencing this issue. @marcosfad can you share the address of the discord channel? Thank you

networkException, 2021-06-18:
This is a link to the server https://discord.com/invite/68MRhQu

AdityaGarg8, 2023-02-10:
Suspend has been finally fixed

closed 2023-02-10 by AdityaGarg8