t2linux/T2-Ubuntu #14: Install not working

dustingooding, 2020-10-07:
I have a 15,1 machine.  I tried installing the latest release (5.8.13) with both `mbp` and `mbp-alt` options.  I can boot into the LiveUSB just fine and poke around and get to the install steps of choosing partitions... but as soon as I click install and it starts going, the screen freezes, the fan spins up for a half-second, and the machine does a hard shutdown.

I can successfully install the previous release (5.7.19) using the `alt` option.

marcosfad, 2020-10-11:
Hi, I've guess, that the problem is the kernel 5.8. Lots of users have been reporting issues in the Discord Channel. There is not much, that I can do here. Sorry.

networkException, 2020-10-28:
Could you add a warning to the readme stating that 5.8.x should be avoided for now? It's usually the main issue when people ask for help on the discord.

reynaldliu, 2021-04-11:
I also encountered this problem. When installing, the fan suddenly speeded up and then it crashed.

I did this
1. Install the system on other computers first
2. Then use squashfs-tools, mksquafs / roo /boot/efi to backup as ubuntu.sqs
3. Start the Apple computer with livecd, mount ubuntu.sqs, and then rsync all files of ubuntu.sqs to the partition to be installed
4. Repair hdx, gptx and UUID in /etc/fstab and grub.cfg to start.

networkException, 2021-04-11:
is there any need for you to use 5.8.x?

AdityaGarg8, 2023-02-10:
Well, we are at 6.1 now :)

closed 2023-02-10 by AdityaGarg8