t2linux/T2-Ubuntu #15: Thunderbolt not working

dustingooding, 2020-10-07:
I have a 15,1 machine, running 5.7.19 `alt` option.  I've got the WiFi and pulseaudio tweaks implemented, but otherwise stock.

I have a Thunderbolt "dock" that has downstream USB devices (keyboard and mouse).  This configuration works fine in MacOS.  In mbp-ubuntu, the Thunderbolt "dock" is detected (in Privacy>Thunderbolt) and is set to "Authorized".  But, my USB devices aren't recognized.  What's interesting is that power is transmitted, and my two external monitors plugged into the dock work fine.  When I use a normal USB hub, the keyboard and mouse work fine.

marcosfad, 2020-10-11:
Wow, this is a hard one to reproduce. 
I'm using a thunderbolt disk where I've install Ubuntu, but that is not the same probably as what you are reporting.
Maybe you can join the discord channel, there you will find great people willing to help out when they can.

cjprinse, 2021-01-14:
I have the same issue with my LG ultrwide thunderbold monitor. The monitor is working fine, but any peripherals are not working. (The monitor is authorized). Is there any news in this issue?

tpchuckles, 2021-07-22:
not sure if it's related, but i've got an apple 27" thunderbolt display, so with the apple usb-c to thunderbolt adapter, it works on macOS, but not on ubuntu 20,04 (5.13-mbp-16x-wifi). 

when i connect and disconnect it, it registers in dmesg, but doesn't work. i'm including my dmesg output below

```[  130.444622] thunderbolt 1-3: new device found, vendor=0x1 device=0x8002
[  130.444640] thunderbolt 1-3: Apple, Inc. Thunderbolt Display
[  133.347855] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
[  137.411467] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
[  137.411545] usb usb2-port4: attempt power cycle
[  141.548319] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
[  145.615040] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
[  145.615082] usb usb2-port4: unable to enumerate USB device
[  299.779178] thunderbolt 1-3: device disconnected
[  308.749472] thunderbolt 1-3: new device found, vendor=0x1 device=0x8002
[  308.749482] thunderbolt 1-3: Apple, Inc. Thunderbolt Display
[  311.695523] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
[  315.758967] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
[  315.759040] usb usb2-port4: attempt power cycle
[  319.892518] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
[  323.943956] apple-ib-touchbar 0003:05AC:8102.0008: hid_field_extract() called with n (64) > 32! (irq/161-bce_dma)
[  323.952582] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
[  323.952655] usb usb2-port4: unable to enumerate USB device```

AdityaGarg8, 2023-02-10:
https://github.com/t2linux/T2-Ubuntu/issues/71

The comment here works for many.

closed 2023-02-10 by AdityaGarg8