t2linux/T2-Ubuntu #16: Docker doesn't work

dustingooding, 2020-10-07:
I have a 15,1 machine, running 5.7.19 `alt` option.  I would like to use Docker, but the kernel isn't configured properly for Docker's needs.  It needs `cgroups`.   Following standard Docker install instructions found here https://docs.docker.com/engine/install/ubuntu/

```
$ uname -a
Linux sneaky-penguin 5.7.19-mbp-alt #1 SMP PREEMPT Sat Aug 29 10:05:12 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

This is the error.

```
$ sudo dockerd
INFO[2020-10-07T16:57:19.930518745-05:00] Starting up                                  
INFO[2020-10-07T16:57:19.931281426-05:00] parsed scheme: "unix"                         module=grpc
INFO[2020-10-07T16:57:19.931293054-05:00] scheme "unix" not registered, fallback to default scheme  module=grpc
INFO[2020-10-07T16:57:19.931307401-05:00] ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock 0  <nil>}] <nil>}  module=grpc
INFO[2020-10-07T16:57:19.931313172-05:00] ClientConn switching balancer to "pick_first"  module=grpc
INFO[2020-10-07T16:57:19.932528156-05:00] parsed scheme: "unix"                         module=grpc
INFO[2020-10-07T16:57:19.932557463-05:00] scheme "unix" not registered, fallback to default scheme  module=grpc
INFO[2020-10-07T16:57:19.932583734-05:00] ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock 0  <nil>}] <nil>}  module=grpc
INFO[2020-10-07T16:57:19.932595631-05:00] ClientConn switching balancer to "pick_first"  module=grpc
INFO[2020-10-07T16:57:19.945075192-05:00] [graphdriver] using prior storage driver: overlay2 
WARN[2020-10-07T16:57:19.964082055-05:00] Your kernel does not support cgroup rt period 
WARN[2020-10-07T16:57:19.964096123-05:00] Your kernel does not support cgroup rt runtime 
WARN[2020-10-07T16:57:19.964101583-05:00] Your kernel does not support cgroup blkio weight 
WARN[2020-10-07T16:57:19.964118062-05:00] Your kernel does not support cgroup blkio weight_device 
INFO[2020-10-07T16:57:19.964226561-05:00] Loading containers: start.                   
INFO[2020-10-07T16:57:19.992694336-05:00] stopping event stream following graceful shutdown  error="<nil>" module=libcontainerd namespace=moby
failed to start daemon: Error initializing network controller: list bridge addresses failed: PredefinedLocalScopeDefaultNetworks List: [172.17.0.0/16 172.18.0.0/16 172.19.0.0/16 172.20.0.0/16 172.21.0.0/16 172.22.0.0/16 172.23.0.0/16 172.24.0.0/16 172.25.0.0/16 172.26.0.0/16 172.27.0.0/16 172.28.0.0/16 172.29.0.0/16 172.30.0.0/16 172.31.0.0/16 192.168.0.0/20 192.168.16.0/20 192.168.32.0/20 192.168.48.0/20 192.168.64.0/20 192.168.80.0/20 192.168.96.0/20 192.168.112.0/20 192.168.128.0/20 192.168.144.0/20 192.168.160.0/20 192.168.176.0/20 192.168.192.0/20 192.168.208.0/20 192.168.224.0/20 192.168.240.0/20]: no available network
```

marcosfad, 2020-10-11:
Hi, I'm not able to reproduce the issue.

```
❯ uname -r
5.7.19-mbp-alt
❯ docker -v
Docker version 19.03.13, build 4484c46d9d
❯ sudo docker run hello-world 

Hello from Docker!
This message shows that your installation appears to be working correctly.

To generate this message, Docker took the following steps:
 1. The Docker client contacted the Docker daemon.
 2. The Docker daemon pulled the "hello-world" image from the Docker Hub.
    (amd64)
 3. The Docker daemon created a new container from that image which runs the
    executable that produces the output you are currently reading.
 4. The Docker daemon streamed that output to the Docker client, which sent it
    to your terminal.

To try something more ambitious, you can run an Ubuntu container with:
 $ docker run -it ubuntu bash

Share images, automate workflows, and more with a free Docker ID:
 https://hub.docker.com/

For more examples and ideas, visit:
 https://docs.docker.com/get-started/
```
CGROUP is enable in the kernel:
```
# cat /boot/config-5.7.19-mbp-alt|grep CGROUP
CONFIG_CGROUPS=y
CONFIG_BLK_CGROUP=y
CONFIG_CGROUP_WRITEBACK=y
CONFIG_CGROUP_SCHED=y
CONFIG_CGROUP_PIDS=y
CONFIG_CGROUP_RDMA=y
CONFIG_CGROUP_FREEZER=y
CONFIG_CGROUP_HUGETLB=y
CONFIG_CGROUP_DEVICE=y
CONFIG_CGROUP_CPUACCT=y
CONFIG_CGROUP_PERF=y
CONFIG_CGROUP_BPF=y
# CONFIG_CGROUP_DEBUG is not set
CONFIG_SOCK_CGROUP_DATA=y
CONFIG_BLK_CGROUP_RWSTAT=y
CONFIG_BLK_CGROUP_IOLATENCY=y
CONFIG_BLK_CGROUP_IOCOST=y
# CONFIG_BFQ_CGROUP_DEBUG is not set
CONFIG_NETFILTER_XT_MATCH_CGROUP=m
CONFIG_NET_CLS_CGROUP=m
CONFIG_CGROUP_NET_PRIO=y
CONFIG_CGROUP_NET_CLASSID=y
```

dustingooding, 2020-11-13:
Ok.  I used MacOS for a few weeks.  I came back today, booted into Linux, and Docker came up just fine.   /shrug

closed 2020-11-13 by dustingooding