t2linux/T2-Ubuntu #18: How to partition in macos Big Sur

Uffje, 2020-10-12:
Does someone know how to partition a disk in Big Sur? For me it doesn't create a partition, but adds a volume to a container disk.

vinodhmoodley, 2020-10-14:
[image]

1. Open Disk Utility and select the Show all Devices option. That will then show all your physical drives in the left column and not just the partitions/volumes.

2. Select the physical drive you wish to partition and then select the Partition option.

closed 2020-10-14 by marcosfad

Uffje, 2020-10-28:
@marcosfad For some reason I can't change the size of the disk or do I need to add a volume first? When I do add a volume it adds it to the container disk1.

[image]

Redecorating, 2020-10-28:
Hi,
The APFS container is currently unshrinkable due to snapshots of the APFS disk taking up the remaining space. I know of two ways to clear them, which will allow you to partition the disk (Making the minimum size diskutility mentions the same as the used space)

1. If you have not installed Windows via bootcamp (from your screenshot it looks like you haven't), you can open the bootcamp installer and follow it up too after it clears snapshots and cached iCloud data to make room for windows. Once it finishes that and asks for a windows iso, you can quit it and go back to disk utility to partition the disk and use the space that was cleared for windows for your Linux install.
2. If you have installed windows, bootcamp won't do this, and you'll have to remove the snapshots manually. You can list them in terminal with this command:
```shellsession
$ tmutil listlocalsnapshots /
Snapshots for volume group containing disk /:
com.apple.TimeMachine.2020-10-27-191838.local
com.apple.TimeMachine.2020-10-27-201730.local
com.apple.TimeMachine.2020-10-28-165152.local
com.apple.TimeMachine.2020-10-28-171735.local
```
and you can delete them individually like this:
```shellsession
$ tmutil deletelocalsnapshots 2020-10-17-121515
```
This command should delete them all without need for copy and pasting the snapshot dates if you need it done automatically (but I haven't tested it)
```
bash -c 'for i in $(tmutil listlocalsnapshots /|grep com);do tmutil deletelocalsnapshots $(echo $i|cut -d "." -f 4);done'
```
I've found that no space is cleared up until all snapshots have been deleted. Once you have deleted them, you can partition the APFS container, and make a partition for linux.

Uffje, 2020-10-28:
Thank you @Redecorating I'll have a look at it tomorrow! The first one sounds like a good plan.