t2linux/T2-Ubuntu #19: Enabling the grub menu

mahboobkarimian, 2020-10-23:
As I am using Docker a lot (so I need mbp kernel linux headers), I tried to install the new kernel (5.8.16-mbp) because the old one (5.7.15-mbp) is disappeared from mbp-ubuntu-kernel.herokuapp.com. After reboot, I had a problem that some people also reported with Macbook Pro 2019.
Then I had a hard time to boot with live-cd to do grub-update since there was not any grub menu and hence Advanced recovery entries to boot at least with the older kernel.
I think enabling the grub menu by default is very helpful in these situations. Especially for us as we need to update to new kernels sometimes and there is no guarantee to having not dead Linux.

* For the people who want to have a menu, they can modify `/etc/default/grub` with the following changes:
```
GRUB_TIMEOUT_STYLE=menu
GRUB_TIMEOUT=10
```
and then:
`sudo update-grub`

marcosfad, 2020-11-06:
Hi, thank you for noticing. I will add you comment to the readme and close this issue.

closed 2020-11-06 by marcosfad

reynaldliu, 2021-04-11:
I also encountered this problem. When installing, the fan suddenly speeded up and then it crashed.

I did this
1. Install the system on other computers first
2. Then use squashfs-tools, mksquafs / roo /boot/efi to backup as ubuntu.sqs
3. Start the Apple computer with livecd, mount ubuntu.sqs, and then rsync all files of ubuntu.sqs to the partition to be installed
4. Repair hdx, gptx and UUID in /etc/fstab and grub.cfg to start.