t2linux/T2-Ubuntu #20: No EFI disk after installation finished

Uffje, 2020-10-29:
After the installing process the fans spin up and the laptop shuts down, then when removing the usb and getting in to the booting screen only Macintosh HD shows up.

Is Ubuntu not installed at all or did I not manage it to show up in the boot screen?

marcosfad, 2020-11-06:
Hi @Uffje, can you provide more information about your Hardware and how did you configure the installer? I think you would profit from posting you problem in the discord channel. There you will find several members willing to help you step by step. It is a great community.

Uffje, 2020-11-06:
@marcosfad I will do that, and close this issue then.

AdityaGarg8, 2023-02-10:
Haven’t received a reply yet :(

closed 2023-02-10 by AdityaGarg8