t2linux/T2-Ubuntu #21: Keyboard and trackpad

AlexCHAUVET34, 2020-11-12:
hello, i installed ubuntu alt on a macbook air 13 2020, but i got kernel panic when i upgrade so i decided to install Kali linux, the problem is the keyboard and the trackpad don't work, i must to use a usb keyboard and mouse to use it, 
Please does anyone can help me to fix it.

thank you very much

marcosfad, 2020-11-12:
Hi, Sorry to hear, that Ubuntu didn't work. If you are interested, you can ask for help in the discord channel. There you will find great people willing to help.
About Kali. You will need to patch the kernel and install the additional modules. Check my other repo https://github.com/marcosfad/mbp-ubuntu-kernel there you will see, what you need to do.
After you patch the kernel, you will need to add some drivers. Take a look at https://github.com/marcosfad/mbp-ubuntu/blob/master/files/chroot_build.sh#L101

I'm not sure, but I think somebody in discord was working with Kali Linux. Maybe you can ask around. 

Hope you are able to get it to work.

AlexCHAUVET34, 2020-11-12:
First, i really want to thank you  to help me and to advice me, finally i can use the keyboard, the trackpad, the sound and the webcam thanks to you, i have installed "linux-headers-5.8.13-mbp_5.8.13-1_amd64.deb" and "linux-image-5.8.13-mbp_5.8.13-1_amd64.deb ".

Now i would like to fix the trackpad to use the right clic and try to fix the wifi.

Thank you for all your help.

marcosfad, 2020-11-13:
Hi, take a look in Discord:
https://discord.com/channels/595304521857630254/722427464604844122/775619491563700225 with the usb id not being in the driver?
(Thanks @Redecorating)

AlexCHAUVET34, 2020-11-13:
Ok, thank you

thalesmaoa, 2022-04-07:
Did Kali boot normally? I was trying to install elementaryOS without success.

acheong08, 2022-04-07:
This repo is for Ubuntu only

EmreTech, 2022-04-07:
> This repo is for Ubuntu only

Mmmmmmm no
You can use the mbp-ubuntu-kernel repo for Debian based distros

acheong08, 2022-04-07:
Check out https://github.com/t2linux for other distros

AdityaGarg8, 2023-02-10:
https://wiki.t2linux.org/distributions/debian/installation/

This documentation is for other Debian/Ubuntu based distros.

closed 2023-02-10 by AdityaGarg8