t2linux/T2-Ubuntu #25: ERROR (dkms apport): binary package for apple-ibridge: 0.1 not found

KeepMasterBranch, 2020-12-11:
Trying to build using current master 4b01e2de56cd807d5be547b2ebea1232cb2b0593:

```
Cloning into '/usr/src/apple-bce-0.1'...
remote: Enumerating objects: 112, done.
remote: Counting objects: 100% (112/112), done.
remote: Compressing objects: 100% (64/64), done.
remote: Total 871 (delta 64), reused 86 (delta 48), pack-reused 759
Receiving objects: 100% (871/871), 193.31 KiB | 568.00 KiB/s, done.
Resolving deltas: 100% (586/586), done.
Note: switching to '85347e71dd79e0be486a79af36862c96027c0836'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 85347e7 Add dkms.conf

Creating symlink /var/lib/dkms/apple-bce/0.1/source ->
                 /usr/src/apple-bce-0.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j12 KERNELRELEASE=5.9.6-mbp...
cleaning build area...(bad exit status: 2)

DKMS: build completed.

apple-bce.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.9.6-mbp/updates/dkms/

depmod.....

Warning: Unable to find an initial ram disk that I know how to handle.
Will not try to make an initrd.

DKMS: install completed.
Cloning into '/usr/src/apple-ibridge-0.1'...
remote: Enumerating objects: 181, done.
remote: Counting objects: 100% (181/181), done.
remote: Compressing objects: 100% (90/90), done.
remote: Total 913 (delta 98), reused 101 (delta 91), pack-reused 732
Receiving objects: 100% (913/913), 391.32 KiB | 758.00 KiB/s, done.
Resolving deltas: 100% (574/574), done.
Note: switching to '90cea3e8e32db60147df8d39836bd1d2a5161871'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 90cea3e appletb: fix inverted logic in detecting internal keyboard/trackpad.

Creating symlink /var/lib/dkms/apple-ibridge/0.1/source ->
                 /usr/src/apple-ibridge-0.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j12 KERNELRELEASE=5.9.6-mbp...(bad exit status: 2)
ERROR (dkms apport): binary package for apple-ibridge: 0.1 not found
Error! Bad return status for module build on kernel: 5.9.6-mbp (x86_64)
Consult /var/lib/dkms/apple-ibridge/0.1/build/make.log for more information.

```

tpchuckles, 2021-07-20:
I saw the same issue ("binary package for apple-ibridge: 0.1 not found") when trying to follow the "from scratch" dkms instructions [here](https://github.com/marcosfad/mbp-ubuntu/#update-to-newer-kernels).

starting with the latest version in the iso (https://github.com/marcosfad/mbp-ubuntu/releases/tag/v20.04-5.7.19-1), i was missing wifi, so i followed these instructions for extracting my macOS firmware (https://wiki.t2linux.org/guides/wifi/), but it still didn't work. I then snatched up the latest custom-built kernel their page linked to (5.13.0 https://github.com/Redecorating/mbp-ubuntu-kernel/releases), at which point i got wifi but lost mouse and keyboard. tried the dkms instructions, and got this error (the make.log said something about " linux/input-polldev.h: no such file or directory"). tried again with an older kernel (5.10.47), which gave me the same message, but a different error in the make.log). finally tried kernel 5.12.13 and both wifi/mouse/keyboard worked. 

I unfortunately don't have the dkms make.log files for the failures: googling their contents wasn't too helpful, and i was more interested in just getting it all working (trying the different kernel which seemed to resolve it)

Redecorating, 2021-07-20:
Try using these instructions for dkms: https://wiki.t2linux.org/guides/dkms/

tpchuckles, 2021-07-20:
> Try using these instructions for dkms: https://wiki.t2linux.org/guides/dkms/

beautiful, thank you!

it looks like the mb15 branch of roadrunner2's apple-ibridge is no longer the latest and greatest. wiki could maybe be updated if that's the case (and i guess this issue isn't technically an issue with marcosfad/mbp-ubuntu either)

AdityaGarg8, 2023-02-10:
The DKMS drivers are now integrated in the kernel.

closed 2023-02-10 by AdityaGarg8