t2linux/T2-Ubuntu #27: USB audio doesn't work

skaba, 2021-01-19:
I have a type-c usb hub attached to my mac and I'm attaching my USB headset and also HDMI monitor to it.

With generic kernel audio input/output for both headset and HDMI screen works
With MBP (tried 5.7.x and 5.8.x) none works. The only difference I can see is dmesg output when I unplug and plug the headset

MBP kernel:
[  206.728913] usb 1-2.3: USB disconnect, device number 9
[  211.055547] usb 1-2.3: new full-speed USB device number 10 using xhci_hcd
[  211.175517] usb 1-2.3: New USB device found, idVendor=1395, idProduct=0025, bcdDevice= 1.00
[  211.175523] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  211.175526] usb 1-2.3: Product: Sennheiser USB headset
[  211.175529] usb 1-2.3: Manufacturer: Sennheiser Communications 
[  211.200117] input: Sennheiser Communications  Sennheiser USB headset as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3/1-2.3:1.3/0003:1395:0025.0010/input/input23
[  211.256080] hid-generic 0003:1395:0025.0010: input,hidraw0: USB HID v1.00 Device [Sennheiser Communications  Sennheiser USB headset] on usb-0000:00:14.0-2.3/input3

Generic kernel:
**[  260.611249] retire_capture_urb: 13 callbacks suppressed
[  260.751901] usb 1-2.3: USB disconnect, device number 4**
[  263.559222] usb 1-2.3: new full-speed USB device number 9 using xhci_hcd
[  263.689925] usb 1-2.3: New USB device found, idVendor=1395, idProduct=0025, bcdDevice= 1.00
[  263.689930] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  263.689934] usb 1-2.3: Product: Sennheiser USB headset
[  263.689937] usb 1-2.3: Manufacturer: Sennheiser Communications 
[  263.716042] input: Sennheiser Communications  Sennheiser USB headset as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3/1-2.3:1.3/0003:1395:0025.0005/input/input17
[  263.775663] hid-generic 0003:1395:0025.0005: input,hidraw0: USB HID v1.00 Device [Sennheiser Communications  Sennheiser USB headset] on usb-0000:00:14.0-2.3/input3

Notice the extra lines that I made bold with generic kernel.

If you need any more information just let me know.

skaba, 2021-01-21:
Following the workaround steps in https://gist.github.com/kevineinarsson/8e5e92664f97508277fefef1b8015fba enabled all audio for me (HDMI audio, USB headset, builtin) thanks, should I keep this open until it's fixed out of the box?

marcosfad, 2021-01-21:
Hi, thank you for the feedback. Unfortunately I have to confess, that this issue can be fixed for everybody. If I fix it for some models, I break it for other one. So it is kind of frustrating to get it for everybody right.

closed 2021-01-21 by marcosfad

skaba, 2021-01-21:
Thanks at least I have a fully working Linux system now (got a Macbook by the company)