t2linux/T2-Ubuntu #29: Cannot reboot after installation

XXXTentacion, 2021-02-17:
I have an 8,1 MacBook Air and after installing Ubuntu, booting into it. It works fine (just no WiFi) but when I reboot again. It just shows a blank screen which I can't do anything about

networkException, 2021-02-17:
What kernel version did you use? You can also join the [Linux on T2 Macs (2018+)](https://discord.gg/68MRhQu) Discord Server to ask for help

XXXTentacion, 2021-02-17:
> What kernel version did you use? You can also join the [Linux on T2 Macs (2018+)](https://discord.gg/68MRhQu) Discord Server to ask for help

I'm using the kernel version 5.8.13

networkException, 2021-02-17:
That might very well be at least some part of the issue, see #26

marcosfad, 2021-09-11:
Hi, if you still experience this issue, can you please try `high | sudo tee /sys/bus/pci/drivers/amdgpu/0000:??:??.?/power_dpm_force_performance_level`. Please replace the ???? with the value you find in lspci. This solved the issue on my macbook pro.
On the next release I'm setting this value as default.

closed 2021-09-17 by XXXTentacion