t2linux/T2-Ubuntu #32: macbook pro 16,1 16inch 2019 wifi helper.

reynaldliu, 2021-04-07:
can you support 16,1 wifi 
i think this may be  can help you 
https://github.com/thomas-jacobsen/mbp16-wifi-driver
https://github.com/jamlam/mbp-16.1-linux-wifi

EmreTech, 2021-04-07:
They are experimental, and even though they work, those drivers shouldn’t be put in yet.

drst3in, 2021-04-09:
I'm using mbp 16,1, and I compiled kernel-5.10.28 using mbp-16.1-linux-wifi, but the wifi doesn't work. What's the problem?

networkException, 2021-04-09:
Depending on what instructions you followed for firmware installation you might need to use a different one. I recommend to look at [this guide](http://wiki.t2linux.org/guides/wifi/#custom-kernel-for-161)

reynaldliu, 2021-04-11:
> Depending on what instructions you followed for firmware installation you might need to use a different one. I recommend to look at [this guide](https://wiki.t2linux.org/guides/wifi/#custom-kernel-for-161)

thank you very mush,i had sucessful use your guide

mahboobkarimian, 2021-04-11:
Yes. This is working in mbp16.1 and I am very happy. I think we need to find a good and easy way to maintain the patches with new kernels.

drst3in, 2021-04-12:
I compiled the kernel again and it was successful, but apple-ibridge does not work... the kernel version is 5.10.28.

when compiling apple-ibridge
/usr/src/apple-ibridge-0.1/apple-ib-als.c:468:28: error: implicit declaration of function ‘iio_priv_to_dev’ [-Werror=implicit-function-declaration]
   468 | struct iio_dev *iio_dev = iio_priv_to_dev(als_dev);

An error occurs like this. What is the problem?

Redecorating, 2021-04-12:
You need a newer version of that kernel module for newer kernels (I think the `iio_priv_to_dev` function was removed). You can get a newer version of the apple-ib-als module by following the instructions [here](https://wiki.t2linux.org/guides/dkms/). If you have issues with your keyboard, trackpad or audio, you may want to also install the version of apple-bce mentioned in that page.

drst3in, 2021-04-12:
> You need a newer version of that kernel module for newer kernels (I think the `iio_priv_to_dev` function was removed). You can get a newer version of the apple-ib-als module by following the [...]

Thank you.
It works

marcosfad, 2021-09-11:
Working on this.

closed 2021-09-30 by reynaldliu