t2linux/T2-Ubuntu #35: Will there be any builds for M1 macs?

TheAlienDrew, 2021-05-14:
Given that the arm64 Mac's are going to be more prevalent in the future....

Redecorating, 2021-05-15:
For linux on m1 macs, you might want to look into [corellium's port](https://corellium.com/blog/linux-m1) or [asahilinux](https://asahilinux.org/), as the m1 macs are very different to intel macs, and don't have the t2 chip.

TheAlienDrew, 2021-05-15:
> For linux on m1 macs, you might want to look into [corellium's port](https://corellium.com/blog/linux-m1) or [asahilinux](https://asahilinux.org/), as the m1 macs are very different to intel macs, and don't have the t2 chip.

@Redecorating very useful info, didn't think about that (brain fart) I'll go ahead and close this.

closed 2021-05-15 by TheAlienDrew