t2linux/T2-Ubuntu #38: fractional scaling....

drst3in, 2021-06-28:
I am using mpb16x, and if I set fractional scaling to 150, my eyes are tired and my head hurts.
100 is fine...
are you all ok?

marcosfad, 2021-09-11:
Hi, I'm using a set up with 2 External Monitors and the laptop screen on the bottom.  I've set fractional scaling to be able to use different scaling in the different monitors. I'm using 150 on mac and 100% in the external monitors. I've not had a problem with this.
But I have a bug, causing that a image of the mouse from the log in screen stays fix on the bottom right of the screen. I've worked that around moving the mouse on log in outside the screen. Have you seen that too?

marcosfad, 2022-02-24:
Too long inactive.

closed 2022-02-24 by marcosfad

AdityaGarg8, 2022-02-25:
It's fixed now anyways