t2linux/T2-Ubuntu #46: Wi-fi trouble in v20.04-5.10.52

aksenovan, 2021-09-10:
I'm trying to install v20.04-5.10.52 ( livecd-5.10.52-mbp-16x-wifi.zip  ) on Macbook:
- MacBook Pro (15-inch, 2019)
- macOS Big Sur 11.5.1
- 2,6 GHz 6x Intel Core i7

After installation as recommended I installed a kernel [v5.13.12-1](https://github.com/AdityaGarg8/T2-Big-Sur-Ubuntu-Kernel/releases/tag/v5.13.12-1) to support wi-fi. But after kernel installation I got no wifi support  as well as a loss of touchbar and keyboard support. 

What could I miss?

AdityaGarg8, 2021-09-11:
@aksenovan , use https://github.com/marcosfad/mbp-ubuntu/releases/tag/v20.04-5.7.19-1.
It's stable and tried and tested.

AdityaGarg8, 2021-09-11:
And follow https://wiki.t2linux.org/guides/dkms/ first

AdityaGarg8, 2021-09-11:
Then upgrade your kernel

AdityaGarg8, 2021-09-11:
And then follow https://wiki.t2linux.org/guides/wifi/

marcosfad, 2021-09-11:
Hi, thank you for reporting your issue. I'm working in a new release which hopefully will fix this kind of problems.

aksenovan, 2021-09-18:
Thanks a lot for your  advice!  v20.04-5.7.19-1 works for me

aksenovan, 2021-09-18:
There is one curious detail: battery charge level in Ubuntu indicates a value less than real (in MacOS). When charge level is ~87% in Ubuntu,  a MacOS one is 100%

AdityaGarg8, 2021-09-19:
That's because of battery optimization by T2. You needn't worry much.

CR0CKER, 2021-10-01:
I'm unable to get wifi to work as well on my MBP 16,2 -- any updated on when a fix may be ready? Looking forward to proceeding with my transition to Linux but can't do it without working wifi and bluetooth. Thanks for your awesome work!

AdityaGarg8, 2021-10-01:
@CR0CKER , may I know the output of uname -r?

CR0CKER, 2021-10-01:
@AdityaGarg8 -- that's 5.10.52-mbp-16x-wifi

AdityaGarg8, 2021-10-01:
@CR0CKER , try https://github.com/AdityaGarg8/T2-Big-Sur-Ubuntu-Kernel/releases/tag/v5.14.9-2

AdityaGarg8, 2021-10-01:
Bluetooth support for 16,2 has been introduced today and is there in this kernel.

AdityaGarg8, 2021-10-01:
For wifi you need to get the correct firmware as described on https://wiki.t2linux.org/guides/wifi/

CR0CKER, 2021-10-01:
Thanks! I'm actually unclear what the instructions on t2linux.org mean -- it says I should "retrieve three of the four files" but I'm not sure how to "retrieve" them (there doesn't seem to be a URL). I was under the impression that downloading and installing the new kernel was enough, but I'm starting to understand that I still need to install the actual drivers separately after downloading them. Thanks for your help on this.

CR0CKER, 2021-10-01:
If there will soon be a new version of mbp-ubuntu that already includes these wifi drivers, I'd rather wait then going though these steps manually.

AdityaGarg8, 2021-10-01:
> If there will soon be a new version of mbp-ubuntu that already includes these wifi drivers, I'd rather wait then going though these steps manually.

These are not the drivers. This is the firmware. It is practically not possible to make a common firmware for all models since each model has its own firmware variety and thus this is the only way out. If you are still having issues @CR0CKER , please contact the discord (https://discord.com/invite/68MRhQu). Someone would help you out through the steps there.

AdityaGarg8, 2021-10-01:
Also, you may contact the discord to get the files retrieved for you.

AdityaGarg8, 2023-02-10:
There have been significant improvements to the wifi, so I don’t think this issue is relevant now.

closed 2023-02-10 by AdityaGarg8