t2linux/T2-Ubuntu #47: Error in step 'Info: Install packages needed for Live System'

yx91490, 2021-09-18:
error log as bellow:

```
===]> Info: Install packages needed for Live System... 
E: Unable to locate package linux-image-5.10.52-mbp
E: Couldn't find any package by glob 'linux-image-5.10.52-mbp'
E: Couldn't find any package by regex 'linux-image-5.10.52-mbp'
E: Unable to locate package linux-headers-5.10.52-mbp
E: Couldn't find any package by glob 'linux-headers-5.10.52-mbp'
E: Couldn't find any package by regex 'linux-headers-5.10.52-mbp'
```
how to solve it?

AdityaGarg8, 2021-09-18:
Do you wish to compile your own kernel?

yx91490, 2021-09-18:
> Do you wish to compile your own kernel?

No, I just want to get an iso format image file.Isnt it the right way?

AdityaGarg8, 2021-09-18:
You can get from the releases

AdityaGarg8, 2021-09-18:
The zip files have the iso files inside

yx91490, 2021-09-19:
> The zip files have the iso files inside

got it

closed 2021-09-19 by yx91490