t2linux/T2-Ubuntu #48: Magenta screen of death?

RaeesBhatti, 2021-09-21:
First of all, thank you so much for creating and maintaining this project. I'm trying to run this on a `MacBookPro15,3` with `BCM4364` WiFi chipset. I tried both `mbp` and `mbp-16x-wifi` versions but I couldn't get it to work.

### `mbp` version:
1. Booted into both Thunderbolt disabled and normal live versions
2. After the boot sequence, it just showed a magenta screen that does funky things when I click:
3. [image]
[image]

### `mbp-16x-wifi` version:
1. Booted into both Thunderbolt disabled and normal live versions
2. After the boot sequence, it loads the normal interface but is non-responsive. The cursor can be moved but nothing happens if I click something.
[image]

I would really appreciate some help with this! Thanks!

RaeesBhatti, 2021-09-21:
Maybe relevant that my system (`MacBookPro15,3`) has Radeon Pro Vega 16 graphics along with Intel UHD Graphics 630

RaeesBhatti, 2021-09-22:
After a bit of debugging, I was able to run it without crashing. It seems like it was the AMD graphics causing the issue. Adding `nomodeset` to boot options fixed the graphics problem. But I wasn't able to make the WiFi work, it seems like I'm having issue described here by some people: https://github.com/mikeeq/mbp-fedora-kernel/issues/3. I followed the [WIFI guide at T2Linux](https://wiki.t2linux.org/guides/wifi/) and copied the firmware files but it still results in
```
brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3  
brcmf_pcie_download_fw_nvram: FW failed to initialize
```

AdityaGarg8, 2021-09-22:
@RaeesBhatti , please use https://github.com/marcosfad/mbp-ubuntu/releases/tag/v20.04-5.7.19-1. Its tried and tested.

RaeesBhatti, 2021-09-22:
Thanks @AdityaGarg8 for the suggestion. Which version should I try, normal or alt?

RaeesBhatti, 2021-09-22:
Also worth mentioning that downloading the firmware from https://packages.aunali1.com/apple/wifi-fw/18G2022/ fixed the WiFi issues.

AdityaGarg8, 2021-09-22:
So no issue left I guess?

RaeesBhatti, 2021-09-22:
I haven't been able to figure out the graphics problem yet. Right now, I'm running with `nomodeset`, which works but that limits it to only internal display. I would like to use both AMD and Intel graphics if that's possible and use external monitors.

closed 2021-09-22 by RaeesBhatti