t2linux/T2-Ubuntu #51: Where is the actual ISO file?

Alex-Fullam, 2022-01-15:
Um, sorry if this is a dumb question. 

Where is the actual iso file? How do I actually use this to get ubuntu working on my Macbook Pro?  Where's the iso?

Runaway1956, 2022-01-15:
https://github.com/marcosfad/mbp-ubuntu/releases/tag/v20.04-20211215214941

You want to download TWO FILES, the .zip and the .z01 and unpack them with 7zip or similar.  Not all archivers will join the two files, but I know 7zip does so.

Alex-Fullam, 2022-01-15:
@Runaway1956 Ok thanks. I'm downloading them on a laptop that already has Ubuntu on it, and I don't have 7zip. I have dowloaded both files (livecd.zip and livecd.z01). What do I do now?

Really sorry if I'm confusing...I'm really confused about this for some reason

Runaway1956, 2022-01-15:
sudo apt install p7zip

And, please read marcosfad README https://github.com/marcosfad/mbp-ubuntu

Alex-Fullam, 2022-01-15:
@Runaway1956 sorry i dont have discord. I have p7zip, but what do I do now? I cannot find any arguments that can join the two files.

Alex-Fullam, 2022-01-15:
@Runaway1956 I have tried many things and cannot figure out how to get p7zip to unpackage them and join the files. Am I doing this right?

Redecorating, 2022-01-16:
you should be able to combine them with `cat livecd.z?? > full.zip` then `unzip full.zip`

Alex-Fullam, 2022-01-16:
@Redecorating This didn't work. The .z01 file is a weird file that when I try to look into the zip it just says it cant display it and also I can't unzip it because of that.

Alex-Fullam, 2022-01-16:
Is everyone sure that I'm supposed to somehow combine the z01 and zip files? Because earlier releases don't even include a z01 file! It's just source code, livecd.zip, and a file called sha256. How am I supposed to get the iso?

AdityaGarg8, 2022-01-16:
@Alex-Fullam 
https://github.com/AdityaGarg8/mbp-ubuntu/releases/tag/v20.04-5.15.14
You can use this ISO. It has everything, including wifi set up already.

AdityaGarg8, 2022-01-16:
Also, marcosfad was bound to split the ISO because the size of the ISO must be greater than 2 GB which is not permitted to be uploaded as a release by GitHub.

vorishirne, 2022-01-25:
@AdityaGarg8 will keyboard and mouse also function out of the box?

AdityaGarg8, 2022-01-25:
> @AdityaGarg8 will keyboard and mouse also function out of the box?

Yes

vorishirne, 2022-01-25:
i am going to use it, but you are a medico, can i rely on this image for my office work?

AdityaGarg8, 2022-01-25:
> i am going to use it, but you are a medico, can i rely on this image for my office work?

It's just a modified @marcosfad  code

AdityaGarg8, 2022-01-25:
You can visit wiki.t2linux.org for more details

vorishirne, 2022-01-25:
ok thanks, seems i can use. But to be brief, what exactly you have modified, just to list

AdityaGarg8, 2022-01-25:
> ok thanks, seems i can use. But to be brief, what exactly you have modified, just to list

Only the kernel. And added T2 specific drivers

AdityaGarg8, 2022-01-25:
> i am going to use it, but you are a medico, can i rely on this image for my office work?

Even I use Linux for Biotech research BTW.

vorishirne, 2022-01-25:
ok thanks

neoh4x0r, 2022-05-24:
@marcosfad It might also be good idea not to include full-paths in the zip or sha256 file.
Using relative paths makes it easier to run **shasum -c sha256** and have it check the file in the current directory.

```
SHA25-HASH filename.iso
```

closed 2023-02-10 by AdityaGarg8