t2linux/T2-Ubuntu #52: MacBook Pro 16, 2019, WIFI does not work.

Zhefan-Xu, 2022-01-18:
Macbook Pro 16 inch, Intel I9 version with Bigsur, does not have WIFI.

I have tried [v20.04-20211215214941](https://github.com/marcosfad/mbp-ubuntu/releases/tag/v20.04-20211215214941), [v20.04-5.10.52](https://github.com/marcosfad/mbp-ubuntu/releases/tag/v20.04-5.10.52) and [v20.04-5.7.19-1](https://github.com/marcosfad/mbp-ubuntu/releases/tag/v20.04-5.7.19-1). But unfortunately, I still cannot use WIFI.

Could please provide any suggestion?
Thanks!

AdityaGarg8, 2022-01-18:
@Zhefan-Xu

AdityaGarg8, 2022-01-18:
Firstly, use the 5.7.19 ISO
Then please upgrade your DKMS drivers and kernel by following https://github.com/AdityaGarg8/T2-Ubuntu-Kernel/blob/Mainline/README.md#pre-installation-steps
The install [this deb](https://cdn.discordapp.com/attachments/706581810745966653/926729551558639646/iso-firmware.deb)
And restart

AdityaGarg8, 2022-01-18:
Wifi should work then

Zhefan-Xu, 2022-01-18:
wow! perfect! IT works

AdityaGarg8, 2022-01-18:
You may close this issue @Zhefan-Xu

closed 2022-01-18 by Zhefan-Xu