t2linux/T2-Ubuntu #53: Issues with changing output device

acheong08, 2022-01-22:
System information:
```
OS: Ubuntu 20.04.3 LTS x86_64 
Host: MacBookPro15,2 1.0 
Kernel: 5.16.2-t2 
Uptime: 4 hours, 27 mins 
Packages: 2009 (dpkg), 7 (snap) 
Shell: zsh 5.8 
Resolution: 2560x1600 
Terminal: gnome-terminal 
CPU: Intel i5-8279U (8) @ 4.100GHz 
GPU: Intel Iris Plus Graphics 655 
Memory: 1964MiB / 7801MiB 
```
What works:
- Built in audio
- WIFI (Fixed by installing third party iwd/focal,now 1.15-1~ppa20.04.1 amd64)
- Touchbar
- Almost everything else
- Can connect bluetooth headphones 

Errors:
- Although the Bluetooth headphones are detected in the Bluetooth menu as an audio device, it doesn't show up in sound settings
- Wired headphones plugged in through the headphone jack is not detected either
- No Mic (Even if I connect an external mic, it is not detected)

AdityaGarg8, 2022-01-22:
You need to manually change sound source from built-in output to codec output in settings>sounds to get the wired headphones working.
External USB mic works for me. Are you using a USB Mic as well?

acheong08, 2022-01-22:
> You need to manually change sound source from built-in output to codec output in settings>sounds to get the wired headphones working. External USB mic works for me. Are you using a USB Mic as well?

In settings>sounds, the only output device shown is 'Multichannel'

acheong08, 2022-01-22:
The mic I am using goes through the headphone jack

AdityaGarg8, 2022-01-22:
https://wiki.t2linux.org/guides/audio-config/

You need to configure sound by following this guide

USB mic works for now.

acheong08, 2022-01-22:
> https://wiki.t2linux.org/guides/audio-config/
> 
> You need to configure sound by following this guide
> 
> USB mic works for now.

I have configured the sound using the guide. No changes.

AdityaGarg8, 2022-01-22:
Did you restart after configuring?

acheong08, 2022-01-22:
Thanks. Fixed. Restarting three times seemed to do the trick. Perhaps this might be a localized issue but sometimes when I reboot from Terminal, it gets stuck and doesn't actually reboot, the screen just turns dark and logs out.

closed 2022-01-22 by acheong08