t2linux/T2-Ubuntu #55: Booting from the Startup Manager results in black screen hang

rabarar, 2022-01-26:
When booting from reFind, Ubuntu boots as expected. Attempting a boot from the Start Up manager (option-boot) selecting the EFI boot results in a black screen hang

AdityaGarg8, 2022-01-26:
[https://cdn.discordapp.com/attachments/706581810745966653/935875060718272592/IMG_20220126_123308.jpg](url)

A pic of the same, from another user on discord facing the same issue.

acheong08, 2022-01-27:
You can partition your disk and install rEFInd and set that as the startup disk. Then set Ubuntu as the default startup in rEFInd.

brad, 2022-02-06:
Any way to make this work without using refind?

AdityaGarg8, 2022-02-07:
https://bugs.launchpad.net/ubuntu/+source/grub2/+bug/1960190

I have reported a bug to Ubuntu.

marcosfad, 2022-02-24:
Hi @rabarar Can you tell me which configuration you chose for your disk partitions on the installer? Can you please also share on which Mac are you installing ubuntu?
I've just installed a fresh ubuntu with the coming release and it worked out of the box (with the configuration detailed in the readme). I think there is a bug on the ubiquity installer which do not allow to use an alternative EFI partition, so it always uses the first EFI partition on the disk.
I would appreciate if you could give me this information so I can try to reproduce the issue.

rabarar, 2022-02-26:
Hi @marcosfad I chose the custom configuration and manual specified each slice to use

I’m installing on a MBP 2019

tmshlvck, 2022-03-23:
I faced apparently the same problem on my MBP 16,2. I am no getting past the "EFI Boot" from Start Up manager (option-boot) to the Grub menu.

I tried both - installing Grub in the MacOS EFI partition (/dev/nvme0n1p1) and creating second EFI partition and installing Grub there manually with grub-install (because installed didn't do it for some reasons). It seems to me that these problems started after I upgraded my Mac to Monterey, but it may be completely wrong assumption and just a coincidence.

AdityaGarg8, 2023-02-10:
As per the current situation looks like this bug is fixed. If you face this issue again, please repoen.

closed 2023-02-10 by AdityaGarg8