t2linux/T2-Ubuntu #56: Only mic is left

vorishirne, 2022-01-26:
Thanks for this wonderful and amazing product. I have made worked everything in my macPro2019 following this issue https://github.com/marcosfad/mbp-ubuntu/issues/52

But mic doesn't works.
With Bluetooth, both audio and mic work but are too bad quality to be used in anyway.

Any way, we can fix or try something? No problem if this is a limitation.

vorishirne, 2022-01-26:
Ok usb mic, and what about the usual audio jack?
Should I get some usual USB to audio jack converter or can u point to some product that you are using

AdityaGarg8, 2022-01-27:
Audio jack just provides the output. Not input afaik. Also, you need to manually change the sound output to audio jack from settings to get the output from there. For input, I am using an ages old Microsoft Lifecam which is actually a webcam but can be used as a Mic. Audio jack to USB port is technically useless.

vorishirne, 2022-01-31:
I got mic working with this https://www.amazon.in/gp/product/B07R4M3WRR/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1

> Audio jack just provides the output. Not input afaik.

However, audio jack is also able to attach mic in idle scenarios

closed 2022-01-31 by vorishirne