t2linux/T2-Ubuntu #57: Shutting lid occasionally results in crash

acheong08, 2022-01-27:
How to reproduce: (Not always)
- Close lid (while charging)
- Unplug charger (With lid closed)
- Plug in charger again
- Open lid
Result:
Crashes and boots to firmware password (EFI boot is not identified as startup disk)
When boot up again, touchbar no longer works
Doing `sudo reboot` does not work (Stuck in shutdown sequence)
To fix touchbar again, must use modprobe to reload the modules then reboot.

AdityaGarg8, 2022-01-27:
Closing the lid cause the Mac to suspend which is broken atm on Linux. You probably can change to closing the lid behaviour to lock instead.

acheong08, 2022-01-27:
Suspend seems to work. If I suspend and turn back on, the screen is completely dark. However, by pressing the power button twice, it turns back on.

AdityaGarg8, 2022-01-27:
What I meant was that it isn’t that stable as you can expect. You are likely to face issues.

closed 2022-01-27 by acheong08