t2linux/T2-Ubuntu #62: Bricked my MacBook Pro

ethanrushh, 2022-03-24:
Not sure what went wrong and I don't think there's a whole lot I can do to fix it now but its bricked my MacBook Pro. 

My setup:

Flasher system: Ubuntu 20.04 LTS, Normal Custom Built PC
MacBook Pro: 256GB, 16GB, i7 MacBook Pro 15,1

Steps I took:

Flashed the ISO onto the USB successfully, booted it from my Mac, let it load, saw the keyboard and touchpad drivers were working then shut it back down gracefully. 

Result:

The Mac no longer boots into anything. Sometimes if I hold option I can get the boot menu but as soon as I click on the Ubuntu USB stick or MacOS it just reboots. Command + R is a no-go as well and just reboots too. The startup chime does sound. 

Not sure how something went this wrong but this needs to be looked into. I didn't modify my disk within the GUI installer in any way, I merely saw it was working then shut it down.

mfadul24, 2022-03-25:
Hi, really sorry to hear. Have you try to restore from internet? https://www.macworld.co.uk/how-to/recovery-mode-internet-3636668/?amp

Did you partition your disk as described on the readme? It sound as you used the whole disk. 

Anyway, if the Mac starts, you can restore it from internet. You will need to be patient though, it may take a while. 

Hope it helps.

marcosfad, 2022-03-25:
Hi, really sorry to hear. Have you try to restore from internet? https://www.macworld.co.uk/how-to/recovery-mode-internet-3636668/?amp
Did you partition your disk as described on the readme? It sound as you used the whole disk.
Anyway, if the Mac starts, you can restore it from internet. You will need to be patient though, it may take a while.
Hope it helps.

ethanrushh, 2022-03-25:
> Hi, really sorry to hear. Have you try to restore from internet? https://www.macworld.co.uk/how-to/recovery-mode-internet-3636668/?amp Did you partition your disk as described on the readme? It [...]

I have tried internet and offline recovery and nada. It's fully dead and not even Apple can help unfortunately. I'm not sure how something went so wrong given that I didn't touch the drive within Ubuntu, maybe a bug in one of the drivers affecting the firmware of one of the chips? 

If there's anything useful I can give you that might help reveal what could have caused this then please let me know, else I think this will have to serve as a caution that it really is unsupported stuff.

AdityaGarg8, 2022-03-26:
@ethanrushh , if you can gain access to another Mac, you can create a bootable macOS installer by following https://support.apple.com/en-us/HT201372 and restore your system back.

ethanrushh, 2022-03-26:
> @ethanrushh , if you can gain access to another Mac, you can create a bootable macOS installer by following https://support.apple.com/en-us/HT201372 and restore your system back.

I can't. It doesn't get that far in the boot sequence. Something goes wrong before it gets to that point. It's like the BIOS is corrupted or something.

AdityaGarg8, 2022-03-26:
So basically you can’t even get into the startup manager by pressing and holding the option key?

AdityaGarg8, 2022-03-26:
Or you are not getting any visual image, including the Apple logo.

AdityaGarg8, 2022-03-26:
If you think it’s a BIOS corruption, then you can revive or restore your Mac using the Apple Configurator utility provided by Apple. Again, you need another Mac to do the same. https://support.apple.com/en-in/guide/apple-configurator-mac/apdebea5be51/mac

AdityaGarg8, 2022-03-26:
> Not getting any image at all but the display does try to initialise (i.e. the backlight comes on) then it reboots.

Then you need this https://support.apple.com/en-in/guide/apple-configurator-mac/apdebea5be51/mac

AdityaGarg8, 2022-03-26:
It will basically reflash the BIOS in the T2.

ethanrushh, 2022-03-26:
> It will basically reflash the BIOS in the T2.

Amazing, I'll give this a go and put the results here shortly. If this works it surely speaks for the instability of the linux drivers...

marcosfad, 2022-03-26:
> If this works it surely speaks for the instability of the linux drivers...
That is only your opinion. 
I’m in the project for now almost 2 years and this is the first time I hear something like this.

ethanrushh, 2022-03-27:
> > If this works it surely speaks for the instability of the linux drivers...
> 
> That is only your opinion. I’m in the project for now almost 2 years and this is the first time I hear something like this.

Nothing opinionated about that. It caused unexpected and unwanted behaviour to a system after proper use. I'm not saying they aren't fantastic (I believe this project is amazing and should be kept going until its perfect) but it clearly isn't perfect and clearly they aren't 100% stable.

ethanrushh, 2022-04-02:
Fixed by binning the Mac. Maybe its crazy coincidence, maybe something is wrong with one of the drivers, I'll never know.

closed 2022-04-02 by ethanrushh