t2linux/T2-Ubuntu #63: Macbook Crashes when attempting to access internal mic

acheong08, 2022-03-28:
I understand that the internal mic is broken and not meant to be used. However, certain applications connect to it as the default before allowing the user to set the default as the external mic. Every time the internal mic is accessed, the computer freezes and i must do a forced reboot to get things working again.

Redecorating, 2022-03-28:
what does `cat /proc/cmdline` say? we may be able to get the mic working which would fix the problem

acheong08, 2022-03-28:
BOOT_IMAGE=/boot/vmlinuz-5.16.10-t2 root=UUID=c7fc442e-2bf8-457a-99cc-4cb56b3eff6d ro efi=noruntime pcie_ports=compat quiet splash vt.handoff=7

Redecorating, 2022-03-28:
https://wiki.t2linux.org/guides/audio-config/#enable-pass-through-kernel-parameters can you follow this section and see if it fixes audio? additionally while you add the iommu ones, you can remove the efi=noruntime, the issue it avoids is fixed by the kernel.

acheong08, 2022-03-28:
fixed. thanks. internal mic now working

closed 2022-03-28 by acheong08