t2linux/T2-Ubuntu #64: Enable multitouch

thalesmaoa, 2022-04-07:
I've installed ubuntu and libgesture to enable multitouch. Do you recommend another route? I can't get pinch to work.

acheong08, 2022-04-07:
Though for Chrome you will need to change the Ozone layer settings in flags

acheong08, 2022-04-07:
For Ubuntu 20.04 LTS you would want to look at touchegg

thalesmaoa, 2022-04-07:
@acheong08, unfortunately, touchegg uses xdotool, which presents some delay issues since my keyboard is US and my language is pt-br. Known bug. Do you suggest to upgrade?

acheong08, 2022-04-07:
I am currently on Ubuntu 21.10 and it works perfectly, though you do have to use apt to install `gnome-shell-extension-desktop-icons-ng` to fix the Desktop. Otherwise desktop icons do not show up.

thalesmaoa, 2022-04-07:
You made it manually or did you use the iso? I don't know how to went from LTS to 21.10.

acheong08, 2022-04-07:
https://askubuntu.com/questions/1369423/upgrade-from-ubuntu-20-04-lts-to-ubuntu-21-10

acheong08, 2022-04-07:
Upgrading might break PPAs. You will have to add them again and reinstall software from said PPAs

closed 2023-02-10 by AdityaGarg8