t2linux/T2-Ubuntu #65: docs for upgrading the kernel?

ldarby-hpe, 2022-04-10:
I installed my system Sept last year and it's working great except for an issue where after several days uptime, unplugging USB devices causes it to reboot, so I want to see if upgrading the kernel resolves this.  I've tried following the docs at https://github.com/marcosfad/mbp-ubuntu#update-to-newer-kernels but can't:

1) I downloaded the livecd.zip v20.04-5.17.1-1, and mounted the squashfs inside but don't see this update_kernel_mbp script anywhere in it (or does the script only get extracted from somewhere once booted from it?).

2) There is nothing relevant about kernel upgrades in https://github.com/marcosfad/mbp-ubuntu/blob/master/files/chroot_build.sh

3) I guess I could try following https://wiki.t2linux.org/guides/kernel/  but I really doubt compiling from source is supposed to be necessary.

Is there supposed to be an apt ppa somewhere?  My system has this:
 # cat /etc/apt/sources.list.d/mbp-ubuntu-kernel.list
deb https://mbp-ubuntu-kernel.herokuapp.com/ /

but there's nothing at that url.

Could you please let me know what the recommended process is and update the docs as such?  Thanks.

ldarby-hpe, 2022-04-10:
Ah, I found the script in the repo, at https://github.com/marcosfad/mbp-ubuntu/blob/master/files/update_kernel_mbp.sh,m and the reason it's missing from the CD is the 01_build_file_system.sh script has this:

cp -r "${ROOT_PATH}/files/update_kernel_mbp.sh" /usr/bin/update_kernel_mbp

which should be  

cp -r "${ROOT_PATH}/files/update_kernel_mbp.sh" "${CHROOT_PATH}"/usr/bin/update_kernel_mbp

ldarby-hpe, 2022-04-10:
Is is really necessary to run that script from the live cd?

Also it references commands dnf and grub2-mkconfig that don't exist (on my system or in the iso, I know dnf is from fedora).

(also I'm not going to run anything until I've understood exactly what needs to happen and what the script is going to do.  I'm extremely worried about issues such as #62 ...)

marcosfad, 2022-04-10:
Hi. Thanks for noticing. I’ve updated the files to now include the script. 
The upgrade process has change a bit since the last release. 
Firstly I stop creating the kernel myself, but started using the one from T2Linux (AdityaGarg8 is doing a great job). 
The last releases of the kernel of T2Linux, do not need any extra patches.
What this means for you:
I guess that you just need to download the release from T2Linux (https://github.com/t2linux/T2-Ubuntu-Kernel/releases/) and before installing it, you will need to remove dkms so the drivers do not override the one on the kernels. 

This should work, but I haven’t tried it. I would recommend you to activate the boot menu in grub, so you can manually select the old kernel if you have problems booting the newer kernel. 

I will try to update the readme to cover this issue. Thank for noticing my mistake on the copy script.

marcosfad, 2022-04-10:
You don’t need to use the script. The issue with the bricked Mac is really strange. To build this iso I’ve had to do lots of stuff to my Mac (at the beginning) and it came back always. 

But of course, as Forrest Gump said: Shit happens.

marcosfad, 2022-04-10:
I’m going to recheck the script in the next weeks. Thanks for noticing the references to dnf

marcosfad, 2022-04-10:
Info about the script

The first part is a self update command. 
Then it looks for the lastest kernel in T2Linux. If it is newer, it download it and install it. 
It reinstall the patch for the suspend issue and re patch grub probe to avoid having problems with the apple partition. 

Then it should clean up and leave only 3 kernels, but this is the part that it is buggy. 

As I said before, you don’t need to use this script, this was just to make it easier.

acheong08, 2022-04-10:
An easier way would downloading the releases from https://github.com/t2linux/T2-Ubuntu-Kernel and installing them.

ldarby-hpe, 2022-04-11:
Thanks, I'll try just downloading the packages later this week, it looks just installing them updates the grub config but I'll double check for this (i.e. if so, then no need to call grub-mkconfig (which exists in ubuntu) in the script).

ldarby-hpe, 2022-04-15:
Just followed https://github.com/t2linux/T2-Ubuntu-Kernel/releases/ (currently v5.15.0-25.25-generic-1 is the latest) and `apt install $HOME/Downloads/*5.15.0-25.25* ` worked fine.  Also I checked the only other thing the script does is add the suspend fix, which I already had.  So I'm not sure the script is necessary at all?  But if you're going to keep it then maybe make it delete /etc/apt/sources.list.d/mbp-ubuntu-kernel.list if that mechanism is deprecated?

AdityaGarg8, 2023-02-10:
The docs for upgrading the kernel are available on https://github.com/t2linux/T2-Ubuntu-Kernel#installation

closed 2023-02-10 by AdityaGarg8