t2linux/T2-Ubuntu #66: Macbook Air 2018 - mic is not working

javacom, 2022-04-12:
When doing zoom, the Macbook Air, microphone is not working.  Audio and webcam are OK.

Redecorating, 2022-04-12:
can you check `cat /proc/cmdline`? If it doesn't contain `iommu=pt` and `intel_iommu=on`, can you follow [this](https://wiki.t2linux.org/guides/audio-config/#enable-pass-through-kernel-parameters), reboot and see if it helps

javacom, 2022-04-14:
following the post I have edited the grub and regenerate then reboot.  But the internal and external microphone are still not working.

acheong08, 2022-04-18:
Are you using Pipewire instead of Pulseaudio? It currently works for me using retaining the Pulseaudio configuration.

AdityaGarg8, 2023-02-10:
There have been updates to the sound configuration files (https://wiki.t2linux.org/guides/audio-config/). Also, its recommended to use pipewire. If you still are facing the issue, please reopen.

closed 2023-02-10 by AdityaGarg8