t2linux/T2-Ubuntu #67: wifi not scanning network

javacom, 2022-04-14:
I used MacBook Air 2018 and there is no wifi settings, I can only connect internet via bluetooth network.
and I do  
`sudo dpkg -i /usr/src/iso-firmware.deb`
The Wifi-Settings does not scan any wifi station for connection.
Then I restart network service  
`sudo systemctl restart NetworkManager.service
 sudo systemctl restart iwd.service`
 
and there are errors as
`-- The job identifier is 8973 and the job result is done.
Apr 14 15:46:10 ubuntu-mpb systemd[1]: iwd.service: Start request repeated too quickly.
Apr 14 15:46:10 ubuntu-mpb systemd[1]: iwd.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- The unit iwd.service has entered the 'failed' state with result 'exit-code'.
Apr 14 15:46:10 ubuntu-mpb systemd[1]: Failed to start Wireless service.
-- Subject: A start job for unit iwd.service has failed
`

AdityaGarg8, 2022-04-14:
May I know what's `uname -r`

AdityaGarg8, 2022-04-14:
I'd also like to have the output of `cat /etc/NetworkManager/NetworkManager.conf` and `cat /etc/NetworkManager/conf.d/wifi_backend.conf`

javacom, 2022-04-14:
uname -r
5.17.1-t2

```
cat /etc/NetworkManager/NetworkManager.conf
[main]
rc-manager=resolvconf
plugins=ifupdown,keyfile
dns=dnsmasq
[ifupdown]
managed=false

```

```
cat /etc/NetworkManager/conf.d/wifi_backend.conf
[device]
wifi.backend=iwd

```

AdityaGarg8, 2022-04-14:
Edit `/etc/NetworkManager/NetworkManager.conf` to look like this

```
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[device]
wifi.scan-rand-mac-address=no
```
 
and edit `/etc/NetworkManager/conf.d/wifi_backend.conf` to look like this

```
#[device]
#wifi.backend=iwd
```

Then run
`sudo systemctl restart NetworkManager`

javacom, 2022-04-14:
Tried and restart service, even reboot still no wifi networks
It is strange that if I do 
`sudo iwlist wlan0 scan`
some ESSID (e.g. my home network 2.4GHz only) can be listed. 

For your information, MacBook Air 2018 is

```
lspci | grep Broad
02:00.0 Network controller: Broadcom Inc. and subsidiaries BCM4355 802.11ac Wireless LAN SoC (rev 0c)

```

AdityaGarg8, 2022-04-14:
> I'd also like to have the output of `cat /etc/NetworkManager/NetworkManager.conf` and `cat /etc/NetworkManager/conf.d/wifi_backend.conf`

Could you send this again so that I make confirm you did the right thing?

javacom, 2022-04-15:
I re-install the ubuntu again and now I plug the USB Wifi Adapter which is 
Bus 001 Device 003: ID 0bda:8172 Realtek Semiconductor Corp. RTL8191SU 802.11n WLAN Adapter

and the wifi  adapter is not scanning at all.
the iwconfig shows
`wlx48022a1a00cd  unassociated  Nickname:"rtl_wifi"
          Mode:Auto  Access Point: Not-Associated   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
`
```
cat /etc/NetworkManager/NetworkManager.conf
[main]
rc-manager=resolvconf
plugins=ifupdown,keyfile
dns=dnsmasq
[ifupdown]
managed=false

```
```
cat /etc/NetworkManager/conf.d/wifi_backend.conf
[device]
wifi.backend=iwd

```

when I changed the NetworkManager.conf and backend.conf to not using iwd
and restart NetworkManager, it works now for the USB wifi Adaptor.

But the internal Broadcom BCM4355 for MacBook Air is still not working.

javacom, 2022-04-15:
OK it is working now for the internal Broadcom BCM4355 for MacBook Air.  
After I 
`sudo systemctl restart NetworkManager
 sudo modprobe brcmfmac`

And then reboot.

closed 2022-04-17 by javacom