t2linux/T2-Ubuntu #69: Ubuntu Live booting freezes

aydar-y, 2022-05-16:
Booting freezes why trying to boot or install Ubuntu from the flashed USB stick (see the screenshot
[image]
)

aydar-y, 2022-05-16:
It's MacBook Pro (16-inch, 2019) Intel

AdityaGarg8, 2022-05-17:
What ISO are you using?

aydar-y, 2022-05-17:
The latest, v20.04-5.17.1-1, ubuntu-20.04.iso downloaded yesterday. The checksum sha256 has also been verified

AdityaGarg8, 2022-05-17:
Could you try this?
https://github.com/AdityaGarg8/mbp-ubuntu/releases/tag/v20.04-5.17.7

aydar-y, 2022-05-18:
Tried it. Same result
[image]
:

aydar-y, 2022-05-18:
diskutil list # found which number has the USB
sudo diskutil umountDisk /dev/diskX
sudo dd bs=4096 if=ubuntu-20.04-XXX.iso of=/dev/diskX****

AdityaGarg8, 2022-05-18:
Could you try using Balena Etcher

AdityaGarg8, 2022-05-18:
Is your macOS updated?
If yes, then try the 5.7.19-mbp ISO from https://github.com/marcosfad/mbp-ubuntu/releases/tag/v20.04-5.7.19-1

Else, first update macOS and then try your USB stick.

aydar-y, 2022-05-20:
It's running macOS Big Sur 11.6.6, the latest of Big Sur. Should I update it to Monterrey?

AdityaGarg8, 2022-05-20:
You may try upgrading to Monterrey.

AdityaGarg8, 2022-05-20:
or first try https://github.com/marcosfad/mbp-ubuntu/releases/tag/v20.04-5.7.19-1

aydar-y, 2022-05-20:
Tried everything, updated to Monterrey- same result everywhere, freezing at boot

AdityaGarg8, 2022-05-20:
Well I made an ISO for 22.04 today.

https://github.com/AdityaGarg8/mbp-ubuntu/releases/tag/v22.04-5.17.9

Maybe it shall work?

Also, @Redecorating , could you help here?

AdityaGarg8, 2022-05-20:
Also, it's possible that it has something to do with your USB stick, cause these ISOs are tried and tested on many MacBook Pro 16 inch models.

aydar-y, 2022-07-11:
None of the images worked. It just freezes at that same point. I tried different USB sticks and adapters.

AdityaGarg8, 2022-07-11:
@Redecorating , could you see the -ENIVAL error shown in the logs

mahboobkarimian, 2022-07-11:
Did You disable the secure boot and do the preparations as described here: https://wiki.t2linux.org/distributions/ubuntu/installation/ ?

aydar-y, 2022-07-11:
Of course I did

closed 2023-02-10 by AdityaGarg8