t2linux/T2-Ubuntu #70: Battery drain to empty on suspended state

M6268, 2022-07-18:
Hi,
I am using Mac book Air 9,1 intel version, installed Ubuntu 22.04 everything works except keyboard back light(as far i am concerned).
I noticed a strange issue one day night i charge to full and suspended and lid closed, next day morning is is not turning on.
Then i connected charger cable it started and showed 1 percent charge.
Next day the same issue.

I have turned off the wifi before suspending.

How can it be fixed.
What data to be shared, how can it be obtained.

acheong08, 2022-07-18:
Suspend on Linux is broken. It is recommended to shut down instead of suspend. When suspended, do not change charge state. For example, if you have your computer suspended and then plug it in/out, there is a high likelihood of the screen being unable to turn on.

Keyboard back-light works on restart. It stops working after suspend.

acheong08, 2022-07-18:
You can change lid close behavior at `/etc/systemd/logind.conf` so that it locks instead of suspend.

acheong08, 2022-07-18:
It is also recommended to update your kernel with https://github.com/t2linux/kernel

acheong08, 2022-07-18:
> What data to be shared, how can it be obtained.

Please provide kernel version and MacBook model. This can be obtained with `neofetch --off`. (`sudo apt install neofetch` if not found)

M6268, 2022-07-18:
Providing neofetch data

OS: Ubuntu 22.04 LTS x86_64 
Host: MacBookAir9,1 1.0 
Kernel: 5.18.8-t2 
Uptime: 9 mins 
Packages: 1932 (dpkg), 25 (flatpak), 6 (snap) 
Shell: bash 5.1.16 
Resolution: 2560x1600 
DE: GNOME 42.2 
WM: Mutter 
WM Theme: Adwaita 
Theme: Yaru-red [GTK2/3] 
Icons: Yaru-red [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i5-1030NG7 (8) @ 400MHz 
GPU: Intel Iris Plus Graphics G7 
Memory: 2308MiB / 7755MiB

AdityaGarg8, 2022-07-18:
Keyboard backlight doesn't work on MacBook Air 9,1
Suspend is currently broken and it's recommended to shut down instead. Currently no fix for the same available.

AdityaGarg8, 2022-07-18:
If you want, you probably can help us in reverse engineering the backlight stuff by joining the discord.

M6268, 2022-07-18:
Sure, share the discord join url.
Any guides to build kernel and test it.
Also why the suspend feature is broken?

AdityaGarg8, 2023-02-10:
As per the current state, both keyboard backlight and suspend have been fixed on the Macs, including MacBookAir9,1.

If you still face any issue, please reopen this issue.

closed 2023-02-10 by AdityaGarg8