t2linux/T2-Ubuntu #72: Error when trying to update from repo

yanivr78, 2023-02-06:
Hi,

Lately there's a new error when trying to update from this repo:
cat /etc/apt/sources.list.d/mbp-ubuntu-kernel.list :
deb https://mbp-ubuntu-kernel.herokuapp.com/ /

Ign:7 https://mbp-ubuntu-kernel.herokuapp.com  InRelease           
Ign:7 https://mbp-ubuntu-kernel.herokuapp.com  InRelease           
Err:7 https://mbp-ubuntu-kernel.herokuapp.com  InRelease
  503  Service Unavailable [IP: 54.73.53.134 443]
Reading package lists... Done
W: Failed to fetch https://mbp-ubuntu-kernel.herokuapp.com/InRelease  503  Service Unavailable [IP: 54.73.53.134 443]
W: Some index files failed to download. They have been ignored, or old ones used instead.

AdityaGarg8, 2023-02-06:
Are you an old user?

If you are a new user, use the ISO from https://github.com/AdityaGarg8/T2-Ubuntu.

marcosfad isn't updating his repos for sometime now. Probably he's busy.

AdityaGarg8, 2023-02-06:
https://github.com/t2linux/T2-Ubuntu-Kernel#pre-installation-steps

And here is the new apt repo, if you are an old user

AdityaGarg8, 2023-02-06:
Remove the marcosfad one from software and updates app.

yanivr78, 2023-02-06:
I can see that under this version that you are suggesting touchbar isn’t supported.Is that true?Cause in the current version that I am using I do have touchbar support.Sent from my iPhone

https://github.com/t2linux/T2-Ubuntu-Kernel#pre-installation-steps

acheong08, 2023-02-06:
> Releases starting from 5.16.2 and 5.15.16 (LTS) have apple-bce and apple-ibridge drivers built-into the kernel. Thus, you may remove the dkms versions of them by running :-

It says that there might be outdated versions of the touchbar drivers. It works without them and you can remove it

AdityaGarg8, 2023-02-06:
@acheong08 is right

yanivr78, 2023-02-06:
I'll try that, Thank you guys!

AdityaGarg8, 2023-02-10:
@yanivr78, in case your issue is resolved, then please close it.

In case of an inactivity of 48hrs, I'll consider the issue as complete.

yanivr78, 2023-02-10:
Thank you! I've re-installed your ISO and its amazing! Many thanks!

closed 2023-02-10 by yanivr78