t2linux/T2-Ubuntu #77: Working on multiple screens

omar-3, 2023-03-09:
The issue is simply that Ubuntu doesn't acknowledge that the MacBook is connected to an external monitor.

AdityaGarg8, 2023-03-10:
Is it a thunderbolt display or HDMI.

omar-3, 2023-03-10:
> Is it a thunderbolt display or HDMI.

HDMI

AdityaGarg8, 2023-03-10:
I actually don't use external monitors but many users on the discord have got it working. You probably could join the server and ask this over there?

omar-3, 2023-03-10:
> I actually don't use external monitors but many users on the discord have got it working. You probably could join the server and ask this over there?

Okay, will close the issue for now.

omar-3, 2023-03-10:
> I actually don't use external monitors but many users on the discord have got it working. You probably could join the server and ask this over there?

Okay, will close the issue for now

AdityaGarg8, 2023-03-10:
I'll repoen it unless we get a solution, so that we can have a record of the same.

lemmyg, 2023-03-11:
Hello,
I use an external monitor with my Macbook  pro 16,1 every day. 
Is connected with HDMI to my usb dock. I have beeen using it in Kubuntu and Ubuntu 22.04, 22.10.
I only had one issue with the laptop HiDPI  screen. where in X11 is not possible to scale each screen separately.
The workaround I found was to set both to 100% and reduce the laptop screen from 3070x1920 to 1920x1200.
Note that Im using amdgpu but intel gpu was working too.
I hope this helps.

lemmyg, 2023-06-19:
note that the hdmi connection required amdgpu.
closing the issue

closed 2023-06-19 by lemmyg