t2linux/T2-Ubuntu #78: rEFInd doesn't see the external drive image

JetForMe, 2023-04-07:
The official 22.04 installer shows up, but your safe graphics one does not. I've tried flashing it twice. Is there anything I can do?

Redecorating, 2023-04-07:
Does the safe graphics installer ISO show up in apple's boot manager (the one you get by holding down the option key when you turn the power on)? It might be an issue with how refind discovers boot options, not an issue with the contents of the USB.

AdityaGarg8, 2023-04-07:
You need to boot the drive using the startup manager.

JetForMe, 2023-04-07:
Ah, interesting. Holding down Option gives three choices: The macOS partition, and two `EFI Boot` options. I don’t know which is which. I chose the first, and after an Apple logo and some screen flashes, a macOS alert appeared saying it needed a software update, with buttons for “Startup Disk…” and “Update.” As this machine has 13.0.1 on it, I decided to let the update happen. We’ll see where this goes.

> You need to boot the drive using the startup manager.

@AdityaGarg8 Is this what you get with Option held down, or is that something else?

JetForMe, 2023-04-07:
So, the second EFI image seemed to be the right one. Sorry to co-opt this ticket, but what is "NVMe Blacklisting?" I can find lots of google results for how to do it, but I don't know *why* I'd want to do that.

AdityaGarg8, 2023-04-08:
> So, the second EFI image seemed to be the right one. Sorry to co-opt this ticket, but what is "NVMe Blacklisting?" I can find lots of google results for how to do it, but I don't know _why_ I'd want to do that.

Some users want to install on an external drive. Blacklisting nvme hides the internal SSD during installation. I did that since the installer is crappy as far as installing on external drives in concerned. Although, not a very effective way since the external drive could also be nvme.

JetForMe, 2023-04-08:
Ah okay. I'm glad I didn't choose that one then, since I was installing on a partition on the internal drive.

closed 2023-04-08 by JetForMe