t2linux/T2-Ubuntu #79: No Trackpad support on MacBookPro16,4 install

JetForMe, 2023-04-07:
I was able to use the keyboard to navigate through the install, but it wasn't until after the installed OS booted and I attached an external mouse that I was able to get mouse control. At that point the trackpad started to work. I did not try attaching an external mouse during the install phase.

AdityaGarg8, 2023-04-08:
Whatever you wrote is too confusing for me. Could you please be more clear.

JetForMe, 2023-04-08:
The trackpad didn't work all throughout the graphical install of Ubuntu. There was no visible pointer and clicking did nothing.

I was able to use the keyboard to navigate through and complete the install.

After that, the trackpad still didn't work.

I plugged in an external mouse, which caused a pointer to appear, and the trackpad to work.

AdityaGarg8, 2023-04-08:
Currently the trackpad works right? Are you able to reproduce the issue on your current installation?

JetForMe, 2023-04-08:
Yes, it works (although not very well, but that's another story). I'm just saying it didn't work until I plugged in a mouse. I haven't tried to reproduce it (a proper repro would require installing Ubuntu all over again.).

AdityaGarg8, 2023-04-09:
That's fine then

closed 2023-04-17 by AdityaGarg8