t2linux/T2-Ubuntu #82: s2idle sleep mode does not work and prevent computer to completely shutdown

ydzhou, 2023-04-30:
Hi

When I use s2idle sleep mode, I run into an issue with my usb controller and I think those errors prevent my computer to completely shutdown or to sleep again

And here is the message during shutdown, then my computer stuck there.
```
Reached target Power-Off
xhci_hcd 0000:08:00.0 host halt failed, -19
```

lsusb output
```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0930:6544 Toshiba Corp. TransMemory-Mini / Kingston DataTraveler 2.0 Stick
Bus 001 Device 006: ID 05ac:8511 Apple, Inc. FaceTime HD Camera (Built-in)
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`0000:08:00.0` is managing Bus 003 and 004

ydzhou, 2023-04-30:
dmesg for suspend/resume
```
[  338.400900] xhci_hcd 0000:08:00.0: Controller not ready at resume -19
[  338.400963] xhci_hcd 0000:08:00.0: PCI post-resume error -19!
[  338.400978] xhci_hcd 0000:08:00.0: HC died; cleaning up
[  338.400989] xhci_hcd 0000:08:00.0: PM: dpm_run_callback(): pci_pm_resume+0x0/0xf0 returns -19
[  338.400994] xhci_hcd 0000:08:00.0: PM: failed to resume: error -19
```

AdityaGarg8, 2023-05-01:
Cc @Redecorating

Redecorating, 2023-05-01:
what mac model is this? has it ever worked in the past (if so, do you know when or what kernel versions)? can you reproduce the issue on the official, not-t2, ubuntu kernel?

Redecorating, 2023-05-01:
also, is normal, not s2idle suspend fine? or maybe if you don't load the xhci_hcd or don't plug in any usb devices, see if it works then?

ydzhou, 2023-05-01:
My kernel is 6.2 (latest t2linux ubuntu kernel)

Redecorating, 2023-05-01:
> The mac is 2019 Macbook Air.
> 
> This is my first time to try t2linux, so it never work in the past. 
> 
> Yes, I can reproduce in non t2 ubuntu kernel.

you might be able to file an upstream bug with ubuntu then.

> 
> Yes, I also tried to not plugin any usb thing, still running into the this issue.
> 
> xhci_hcd is a built-in module. I cannot unload....

ah right 
> 
> Yeah s2idle does not work, but deep mode suspend works.
> 
> 
> 
> BTW, are you able to reproduce s2idle issue on your side?

I haven't tested that, I'm not sure when I'll get time.

ydzhou, 2023-05-01:
Can you do me a favor and try a simple test?
I wanna verify if this is my device issue or this is upstream kernel bug...

```
echo 's2idle' > /sys/power/mem_sleep
systemctl suspend
```
Then wake up
```
dmesg | grep xhci
```

stereo4nick, 2024-07-11:
Hi there! I have updated to sonoma 2020 MBA. Installed Ubuntu 6.9.8-3-t2-jammy and experienced issue with not working suspend/hibernate. I know about same issue on other distros like debian/Arch. 

Recently find out it appeared new workaround based on service which shut down brcmfmac/apple-bce modules before suspend and start them after resume. I tried that with no luck (see https://github.com/t2linux/T2-Debian-and-Ubuntu-Kernel/issues/53#issuecomment-2221615015)

Is there exists some workaround for ubuntu or maybe the issue will be addressed in future? AdityaGarg8 said that he is not working for ubuntu.