t2linux/T2-Ubuntu #83: kubuntu support from lunar version

lemmyg, 2023-05-01:
Hello AdityaGarg,
Im opening  this issue to start talking about how we could modify your repo to also generate the Kubuntu iso.
Let me know what you think and how I can help.
I made a branch with some changes for Kubuntu.
https://github.com/lemmyg/T2-Ubuntu/tree/kubuntuLunar
KDE install fine except plasma-discover that for some reason it gets removed after copy files process. Finally I need to install it manually. Also snap firefox is missing.

Regards,
lemmyg

AdityaGarg8, 2023-05-01:
Hi

Since I am not that aware of the requirements to add support for KDE, I’ve sent you an invite as a collaborator. You can maintainer Kubuntu along with Ubuntu in separate branches and we can release both ISOs together.

lemmyg, 2023-05-01:
that sounds perfect.

lemmyg, 2023-05-01:
how to do want me to rename my branch kubuntuMineline ?

AdityaGarg8, 2023-05-01:
Yeah use LTS and mainline with something as names

lemmyg, 2023-05-04:
So  the iso need to be less than 2GB. Currently is 2.7GB. I not sure if I would be able to reduce that much. Ill try

lemmyg, 2023-05-04:
I did some tests and the smaller iso  is  2.2GB. Replacing ubuntu-standard by kde-standard.
I remember that you tried ones with my kubuntu repo and I think you faced same issue.

AdityaGarg8, 2023-05-04:
Hmm,I guess we shall have to split the ISOs then

lemmyg, 2023-05-04:
are you going to try to modify the CI?

lemmyg, 2023-05-06:
Hello @AdityaGarg8 ,
I was wondering if could have time to resolve the CI this weekend.

AdityaGarg8, 2023-05-06:
> Hello @AdityaGarg8 , I was wondering if could have time to resolve the CI this weekend.

I'll try to get it done this weekend.

lemmyg, 2023-05-06:
That would be great. I'm pretty sure lot of KDE users will like it.

AdityaGarg8, 2023-05-06:
@lemmyg, the latest release has kubuntu ISOs as well now, although they need to be tested.

lemmyg, 2023-05-06:
Nice, I tested those iso with virtualbox.
They install without issues.
Unfortunately still I need to do a backup of my system. We could ask In the forum.
Maybe there is people that they recently installed Ubuntu and don't mind

lemmyg, 2023-05-06:
Ill give a try with virtualbox again.

lemmyg, 2023-05-06:
installed fine. 
user only need to install couple of apps manually
sudo apt install plasma-discover firefox

AdityaGarg8, 2023-05-06:
Nice. I’ll see if I can make an ISO download script to make things easier.

lemmyg, 2023-05-06:
Much better, other ways we need to do
cat myfile.z* > myfilefinal.zip
Unzip myfilefike.zip

lemmyg, 2023-05-12:
closing this issue as the new iso installs without issues.

AdityaGarg8, 2023-05-12:
> Much better, other ways we need to do cat myfile.z* > myfilefinal.zip Unzip myfilefike.zip

I didn't read this earlier, but for some reason `cat myfile.z*` isn't working for me but `zip -F myfile.zip --out myfilefinal.zip` works.