t2linux/T2-Ubuntu #86: Make drive bootable on Windows

redplane, 2023-05-03:
Hi,

Firstly, thank you for your outstanding `T2-Ubuntu`.

I have installed Ubuntu successfully into an **external SSD drive** and able to boot to it from my Macbook 2019 Pro.

However, when I plug the SSD into my Windows PC, it cannot be recognized as a bootable drive.

Any tutorial to make the external SSD to boot on the Windows PC also ? That would be nice because I can use the external drive at my office (Windows PC) and at home (Macbook pro 2019).

Thanks,

AdityaGarg8, 2023-05-04:
Share `lsblk`

redplane, 2023-05-04:
I dont get it, can you clarify more ?

AdityaGarg8, 2023-05-04:
Boot into Ubuntu installed on you external drive using your Mac. Open the terminal there and run the command `lsblk`. It will display an output. Share that here.

redplane, 2023-05-04:
Here is what it printed:
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0  53,2M  1 loop /snap/snapd/19122
loop1         7:1    0  63,3M  1 loop /snap/core20/1852
loop2         7:2    0     4K  1 loop /snap/bare/5
loop3         7:3    0  91,7M  1 loop /snap/gtk-common-themes/1535
loop4         7:4    0 241,5M  1 loop /snap/firefox/2605
loop5         7:5    0 349,7M  1 loop /snap/gnome-3-38-2004/140
sda           8:0    0 465,8G  0 disk 
├─sda1        8:1    0    16G  0 part [SWAP]
├─sda2        8:2    0   449G  0 part /var/snap/firefox/common/host-hunspell
│                                     /
└─sda3        8:3    0 728,5M  0 part /media/sniper/Ubuntu Boot Loader
sdb           8:16   1     0B  0 disk 
sdc           8:32   1     0B  0 disk 
sdd           8:48   1  28,8G  0 disk 
└─sdd1        8:49   1  28,8G  0 part /media/sniper/WIN10PE2021
nvme0n1     259:0    0 465,9G  0 disk 
├─nvme0n1p1 259:1    0   300M  0 part /boot/efi
└─nvme0n1p2 259:2    0 465,6G  0 part 
```

AdityaGarg8, 2023-05-04:
You didn’t create a separate EFI partition for Ubuntu.

AdityaGarg8, 2023-05-04:
I’d like to know did you intend to use sda3 as a separate EFI partition

redplane, 2023-05-04:
> I’d like to know did you intend to use sda3 as a separate EFI partition?

Actually I followed [this tutorial](https://florisvanbreugel.wordpress.com/2018/03/23/installing-ubuntu-on-an-external-ssd-drive-on-a-macbook/) to install normal Ubuntu before trying with `T2-Ubuntu`.

I didn't delete `sda3` before install `T2-Ubuntu`

AdityaGarg8, 2023-05-05:
Rename `Ubuntu Boot Loader partition` to `EFI2` using macOS. Then boot into Ubuntu and run

```
wget https://wiki.t2linux.org/tools/efi.sh
bash ./efi.sh EFI2
```

redplane, 2023-05-06:
> Rename `Ubuntu Boot Loader` partition to `EFI2` using macOS. Then boot into Ubuntu and run
> 
> 
> 
> ```
> 
> wget https://wiki.t2linux.org/tools/efi.sh
> 
> bash ./efi.sh EFI2
> 
> ```

I have tried this but failed with the below message:

./efi.sh: 4: set: Illegal option -o pipefail

The EFI2 drive is hfs+, is it the problem ?

AdityaGarg8, 2023-05-06:
You need to format EFI2 to MS DOS (FAT).

Also share `cat /etc/fstab`

redplane, 2023-05-06:
Still the same.
I have changed `EFI2` to `FAT` as the below screenshot:
[image]

This is the output of `cat /etc/fstab`
```
# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
# / was on /dev/sda2 during installation
UUID=31b7379a-71b4-40b6-92ba-25ee018bd698 /               ext4    errors=remount-ro 0       1
# /boot/efi was on /dev/nvme0n1p1 during installation
UUID=5F66-17ED  /boot/efi       vfat    umask=0077      0       1
# swap was on /dev/sda1 during installation
UUID=0e99845e-16de-4119-ab18-7bc2df5c52cc none            swap    sw              0       0

```

AdityaGarg8, 2023-05-06:
Share `sudo ls /boot/efi/EFI`

redplane, 2023-05-06:
Here you are:

```
APPLE  BOOT  ubuntu
```

AdityaGarg8, 2023-05-06:
Also, if you run `sudo gdisk /dev/sda`, is MBR in output hybrid or something else?

redplane, 2023-05-06:
This is what I got:
```
GPT fdisk (gdisk) version 1.0.8

Problem opening /dev/sda~ for reading! Error is 2.
The specified file does not exist!
```

AdityaGarg8, 2023-05-06:
```
sudo mount /dev/sda3 /mnt
sudo cp -r /boot/efi/EFI /mnt
sudo rm -r /mnt/EFI/APPLE
sudo rm -r /boot/efi/EFI/BOOT
sudo rm -r /boot/efi/EFI/ubuntu
sudo sed -i "s/5F66-17ED/19D7-F9BE/g" /etc/fstab
```

AdityaGarg8, 2023-05-06:
After running the above commands reboot and share 

1. `cat /etc/fstab`
2. `lsblk`

redplane, 2023-05-06:
Will this command delete my MACOS EFI ?
Just want to make sure because I don't want to install MacOS :)
`
sudo rm -r /mnt/EFI/APPLE
`

AdityaGarg8, 2023-05-06:
> Will this command delete my MACOS EFI ? Just want to make sure because I don't want to install MacOS :) `sudo rm -r /mnt/EFI/APPLE`

Don't worry. macOS won't be touched.

redplane, 2023-05-06:
1. Output of `cat /etc/fstab`
```
# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
# / was on /dev/sda2 during installation
UUID=31b7379a-71b4-40b6-92ba-25ee018bd698 /               ext4    errors=remount-ro 0       1
# /boot/efi was on /dev/nvme0n1p1 during installation
UUID=19D7-F9BE  /boot/efi       vfat    umask=0077      0       1
# swap was on /dev/sda1 during installation
UUID=0e99845e-16de-4119-ab18-7bc2df5c52cc none            swap    sw              0       0
```

2. Output of `lsblk`
```
loop0         7:0    0     4K  1 loop /snap/bare/5
loop1         7:1    0  55,6M  1 loop /snap/core18/2721
loop2         7:2    0  63,3M  1 loop /snap/core20/1879
loop3         7:3    0 241,5M  1 loop /snap/firefox/2605
loop4         7:4    0 199,9M  1 loop /snap/gitkraken/207
loop5         7:5    0 349,7M  1 loop /snap/gnome-3-38-2004/140
loop6         7:6    0  91,7M  1 loop /snap/gtk-common-themes/1535
loop7         7:7    0  53,2M  1 loop /snap/snapd/19122
sda           8:0    0 465,8G  0 disk 
├─sda1        8:1    0  15,3G  0 part [SWAP]
├─sda2        8:2    0 448,9G  0 part /var/snap/firefox/common/host-hunspell
│                                     /
└─sda3        8:3    0   1,4G  0 part /boot/efi
sdb           8:16   1     0B  0 disk 
sdc           8:32   1     0B  0 disk 
nvme0n1     259:0    0 465,9G  0 disk 
├─nvme0n1p1 259:1    0   300M  0 part 
└─nvme0n1p2 259:2    0 465,6G  0 part 

```

AdityaGarg8, 2023-05-06:
Looks good now. Try booting using a Windows PC

redplane, 2023-05-06:
Ha ha, thank you for your help.

[image][image]

[image]

Worked on PC and macbook now.
I will summarize the steps that I have taken in the below comment for those who want to make an Ubuntu SSD for PC and Macbook

redplane, 2023-05-06:
By the way, what does this command do ? I want to explain every step
```
sudo sed -i "s/5F66-17ED/19D7-F9BE/g" /etc/fstab
```

AdityaGarg8, 2023-05-06:
> By the way, what does this command do ? I want to explain every step
> 
> ```
> sudo sed -i "s/5F66-17ED/19D7-F9BE/g" /etc/fstab
> ```

We basically changed the EFI partition for Ubuntu from internal drive to external.

5F66-17ED is the UUID of old EFI partition and 19D7-D9BE I'd UUID of new partition.

I got the old UUID from fstab and and new UUID from a screenshot you sent me in a previous chat.

The command replaces the old UUID with new one in the fstab file.

Also, the T2 Ubuntu ISO and a nvme blacklisted option which is basically to make installation on external drives easier. It hides the internal drive from the installer and thus the EFI partition gets created in the external drive only. Basically you didn't follow the correct guide before, this encountered the issue.

redplane, 2023-05-06:
Thanks,

I'll keep this post open and close it later when I summarize the steps, this will help in future use.
I think there should be a proper tutorial for this because it helps much for people who need to use different OS

redplane, 2023-05-06:
With the help of @AdityaGarg8 , I could make this possible, the steps that I have taken was mentioned here: https://superuser.com/questions/1782079/make-existing-external-ubuntu-drive-bootable-on-mac-os/1782680#1782680

Hope this helps.

closed 2023-05-06 by redplane

AdityaGarg8, 2023-05-06:
Hi @redplane , it was a nice answer! But just an advice for EFI partition, you don’t need more than 300Mbs for disk space. So you needn’t waste space by allocating 1Gb to the same.

redplane, 2023-05-06:
> Hi @redplane , it was a nice answer! But just an advice for EFI partition, you don’t need more than 300Mbs for disk space. So you needn’t waste space by allocating 1Gb to the same.
> 
> Also, device for bootloader installation is always an EFI partition, not ext4 as you mentioned there.

Yep,
The steps I mentioned above were the ones I did, so I just share the process :)