t2linux/T2-Ubuntu #88: snd_hda_macbookpro module iMac 19,1

lui75, 2023-05-18:
Hi, why don't you add davidjo patches to your kernel for enabling audio on macbook pro and iMac? I have tried to build and install the modules but the lack of linux-source deb package for your kernel makes it impossible. So, now my wifi is working perfectly but I can not have audio working
Here's the repository: https://github.com/davidjo/snd_hda_macbookpro

AdityaGarg8, 2023-05-18:
In line 84 and 87 of https://github.com/davidjo/snd_hda_macbookpro/blob/master/install.cirrus.driver.sh, change `isubuntu=1` to `isubuntu=0` and run the script again.

lui75, 2023-05-18:
I got this message from T2 Ubuntu kernel:
https://github.com/t2linux/T2-Ubuntu/issues/88#issuecomment-1553479103

lui75, 2023-05-20:
It worked flawlessly! If you decide to add the driver into the kernel, I will be a tester. Let me know what I have to do and how. I am happy to be of help.

AdityaGarg8, 2023-05-20:
If that works then good. Still if you need it in the kernel, reopen this issue.

Also, in case you are using the T2 kernels, make sure the kernel upgrade script is set up to receive the updates.

https://github.com/t2linux/T2-Ubuntu-Kernel#using-the-kernel-upgrade-script

closed 2023-05-20 by AdityaGarg8