t2linux/T2-Ubuntu #89: Disable touchbar, use function keys instead

redplane, 2023-06-17:
Hi,

I have installed `T2-Ubuntu` [into a SSD Drive](https://github.com/t2linux/T2-Ubuntu/issues/86)  use `T2-Ubuntu` for a while.
It is OK on my Macbook Pro 2019, but I don't want to use the touchbar function.
I want to use F1 - F12 keys instead.

Are there any tutorials for disabling the function keys ?
Thanks

amartyadav, 2023-06-17:
You should be able to just press the **fn** key (function key) on your macbook to use the **F1...F12** keys on your touchbar. 
It is a bit inconvenient to do that everytime you want to use the function keys, but it works. 
As of now, I am not aware of a way to replace the touch bar strip with the funtion keys permanently.

AdityaGarg8, 2023-06-18:
Run `sudo touchbar` to configure your touchbar

lemmyg, 2023-06-18:
thats very cool. I didnt know about it

redplane, 2023-06-24:
> Run `sudo touchbar` to configure your touchbar

Hi, I tried this command but it said: command not found

AdityaGarg8, 2023-06-24:
Did you setup the kernel upgrade script? It has the touchbar configuration functionality built in.

If not , run

```
sudo apt update
sudo apt install t2-kernel-script
```

Then try running `sudo touchbar` again.

redplane, 2023-06-26:
Thanks, the `sudo touchbar` command works.
It displayed a menu:
```
0: Only show F1-F12
1: Show media and brightness controls, use the fn key to switch to F1-F12
2: Show F1-F12, use the fn key to switch to media and brightness controls
3: Only show media and brightness controls
4: Only show the escape key
```

When I tried the second option which is `2: Show F1-F12, use the fn key to switch to media and brightness controls`.
There was a message:

```
Changing default mode ...
bash: line 1: /sys/class/input/*/device/fnmode: No such file or directory

```

Anything missing ?

AdityaGarg8, 2023-06-26:
Share the output of `sudo dmesg | grep touchbar`

redplane, 2023-06-26:
It output nothing :(
[image]

AdityaGarg8, 2023-06-26:
Share `lsmod` and `uname -r`

redplane, 2023-06-26:
> sudo apt install t2-kernel-script

I think the problem was because I used the command on my PC, not on my Macbook.
The command ran fine on Macbook

closed 2023-06-26 by AdityaGarg8