t2linux/T2-Ubuntu #90: KUbuntu release assets (zip/z01 files) only available for v23/Lunar, missing for v22/Jammy (iso.sh fails to download)

danielweck, 2023-06-19:
...this seems to be the case for all releases:

https://github.com/t2linux/T2-Ubuntu/releases

...but I also noticed that the latest release `v6.3.8-1` has a duplicate 1.46GB `.z02` file in its assets:

https://github.com/t2linux/T2-Ubuntu/releases/tag/v6.3.8-1

lemmyg, 2023-06-19:
We don't have support for Kubuntu 22.04 at the moment. if you wait for a couple of days. I'll try to add it.

AdityaGarg8, 2023-06-19:
> ...ah, I've just realised that ZIP parts are hard-coded in `iso.sh`:
> 
> https://github.com/t2linux/T2-Ubuntu/blob/c054ccdd1a7ad6a1d8b377244876f2ab8c6ac6bb/.github/workflows/iso.sh#L88-L91
> 
> (still, there seems to be a problem with the KUnbuntu artefacts ... in the meantime I'll be testing with Unbuntu, thank you :)

Indeed there is an issue with Kubuntu's artifacts, I've tried to rectify the issue. Try running iso.sh again after the new artifacts are uploaded in the latest release.

Also, we support kubuntu 23.04 only for now, @lemmyg being our maintainer for Kubuntu. He can help you out for 22.04 LTS.

lemmyg, 2023-06-19:
https://github.com/t2linux/T2-Ubuntu/pull/91

closed 2023-06-21 by danielweck