t2linux/T2-Ubuntu #92: renamed branchs

lemmyg, 2023-06-20:
it would be better if we rename branches:
jammy:
ubuntu-22.04
kubuntu-22.04
lunar:
ubuntu-23.04
kubuntu-23.04

what do you think?

AdityaGarg8, 2023-06-21:
Tbh I was thinking of ubuntu-LTS and ubuntu-Mainline, else well shall have to change the branch names every time a new version of Ubuntu comes, and update the CI accordingly

lemmyg, 2023-06-21:
I dont know   I like to be specific like 22.04. same as when we release the iso. About CI, change it every 6 months is not a big deal.

lemmyg, 2023-06-21:
By the way, when you have spare tiene could you please have a look why the CI is failing?

lemmyg, 2023-06-21:
The good thing of having Ubuntu versions name is that you need create a new
branch every new version and thats allows user build an old version if they
need to

AdityaGarg8, 2023-06-21:
> By the way, when you have spare time, could you please have a look why the CI is failing?

Only the Mainline part is failing, as the Mainline branch already has the version the CI is trying to push, the ISO bit is the one that should succeed. It although was not releasing the iso, which I’ve fixed.

lemmyg, 2023-06-22:
So are we good now?
I hope you saw my comment in chat  about 6.3.9. I may need to skip it

AdityaGarg8, 2023-06-23:
*sigh*, we’ll have to find the patch causing regression now.

lemmyg, 2023-07-08:
HI @AdityaGarg8 ,
I was jammy CI and I don't understand why it mainline and jobs. For example mainlain branch has only iso. Should we remove Kubuntumainline job for kubuntujammy?
Thanks

AdityaGarg8, 2023-07-08:
I didn’t understand what you mean?

lemmyg, 2023-07-08:
Some branches have 2 jobs in CI and others just one.
Why is that?

lemmyg, 2023-07-08:
Just try to find out why Kubuntumainline CI is not working

AdityaGarg8, 2023-07-08:
Like I change the kernel ver

AdityaGarg8, 2023-07-08:
Looks like it closed it by mistake

lemmyg, 2023-07-08:
Cool, I thought something had bothered you.

AdityaGarg8, 2023-07-08:
Btw what's wrong with the CI?

lemmyg, 2023-07-08:
I made to the Kubuntu mainline branch. Still we didn't generate any iso for
it. Last time failed.

AdityaGarg8, 2023-07-08:
Both Kubuntu 22.04 and 23.04 are released in the latest release. Can you check again?

lemmyg, 2023-07-08:
Oh yes sorry.
All good then

closed 2023-07-19 by lemmyg