t2linux/T2-Ubuntu #101: Kubuntu, but no ethernet

Taddaeus, 2023-10-21:
I am very pleased that Kubuntu is now supported!
However, I do not even have Ethernet with my Macbook Pro 13" 2018 touchbar - tried with Kubuntu 23.10 v6.5.7-2 and v6.5.8-1.
Surely this should work directly? Or are the extra steps for WIFI (don't need it and didn't set it up) needed for Ethernet as well?

Or am I missing something?

AdityaGarg8, 2023-10-21:
Does your ethernet adapter work with other Linux PCs?

Taddaeus, 2023-10-21:
Hi
Yes. In addition, I have tried 4 different ones.
However, a connection must be active, because time and date are correct - I did not set them manually.
I tried ping: command not found.
The “TaskManager” does not show any network activity.
Also no audio and no keyboard backlight...

AdityaGarg8, 2023-10-21:
Can you share your `journalctl -k`. You could you a service like pastebin for that

Taddaeus, 2023-10-21:
You can find it here:

 https://privatebin.at/?0c7038f870f5283b#5LVqfoRt4cdKCH8ZNTmpEgpWD6LByKmgfkpNDYbpewCp <https://privatebin.at/?0c7038f870f5283b#5LVqfoRt4cdKCH8ZNTmpEgpWD6LByKmgfkpNDYbpewCp>

PW: Taddaeus

Thanks for investigate

AdityaGarg8, 2023-10-21:
It’s asking for password

Taddaeus, 2023-10-21:
PW: Taddaeus
 
sorry, should have done without password..

closed 2024-03-10 by AdityaGarg8