t2linux/T2-Ubuntu #103: Can't install the latest release

Rom888, 2023-10-23:
I tried to install the latest version:
ubuntu-23.10-6.5.8-t2-mantic.zip

After selecting "Install" or "Try" I only see a flashing screen with:
```
BCMDHD_ExitbootService: Enter
dhd_os_wifi_platform...
dhd_os_wifi_platform...
```
pc:
```
macbook pro, 16 inch, 2019
model identifier: MacBookPro16.1
Intel Core i9 2.3g
Intel UHD graphics 630
AMD radeon pro 5500M
```

AdityaGarg8, 2023-10-23:
Which software are you using to burn the iso?

AdityaGarg8, 2023-10-23:
Try using balena etcher

Rom888, 2023-10-23:
Thank you, tried, but without success 
[image]

Rom888, 2023-10-23:
I have installed apple-set-os loader.
Could this be the reason?

AdityaGarg8, 2023-10-23:
Seems to be some issue with your USB

Rom888, 2023-10-23:
I tried with another USB, and the same result.
Also the Ubuntu Startup Disk Creator doesn't accept this ISO.

AdityaGarg8, 2023-10-23:
Try burning using macOS

Rom888, 2023-10-23:
After rebooting into macOS everything is fine.
It looks like it was because
nvram fa4ce28d-b62f-4c99-9cc3-6815686e30f9:gpu-power-prefs=%01%00%00%00

AdityaGarg8, 2023-10-23:
Possible

closed 2023-10-23 by AdityaGarg8

Sprunglicious, 2023-10-26:
I have experienced the issue in the past. Use the Ventoy ("https://ventoy.net/en/download.html") utility. I have never had an issue since I begun to use the utility.